ktechlab/ktechlab #29: Clock input not working well at all

BenjiWiebe, 2015-09-15:
I'm not sure why, but the clock input is not only very far away from the speed I set, but it is also extremely erratic. When set at 50Hz, it runs at maybe 5Hz for 3 seconds, and then suddenly sticks On for about 10 seconds, then goes at 1Hz for a couple seconds. This is very annoying -- I'd rather it always went 1/10th of the set speed or something rather than be so erratic.

zoltanp, 2015-09-16:
Hi BenjiWiebe,

I assume you are using the current KDE4 port of KTechLab 0.3.

I have a few questions related to the issue:
* How much CPU usage does ktechlab do, when you use the 50Hz Clock Input? And is there any CPU power management policy set on your system?
I expect that ktechlab uses a lot of CPU time, but hopefully it doesn't force the CPU to generate too much heat and make the CPU to enter thermal protection -- read: downclocking -- region. Normally this should not happen, but I consider that it is good to be verified. Also the CPU frequency governor might make the CPU to jump between operating frequencies, and cause strange timings in ktechlab.

* In the current implementation, ktechlab 
  * refreshes the display at 50 Hz by default (as far as I remember). This is configurable.
  * runs the simulation loop 50 times a second
  * in each simulation loop, it runs 200 simulation steps
  * in each simulation step it might change the state of the Clock Input.

So the actual output signal of the clock input at 50 Hz is very heavily undersampled by the display, and any rounding error in time calculation this might exhibit the behavior you are describing. Still, in that case the actual signal generated by clock input should be correct -- this can be verified by connecting the clock input to a divider.

I will verify if the Clock Input's code is "robust" and also I will verify that it outputs the expected signal.

BenjiWiebe, 2015-09-16:
@zoltanp I tried forcing the CPU speed to 4Ghz, and the issue persisted (and CPU usage was ~89% of one core, and the temperature was 38.0°C). Also, with the clock input hooked up to a counter, the counter was counting the pulses I could see -- that is, the output of the counter would increment steadily and then suddenly increment by 4 or 5 far faster than it had been going, and then revert to the slow speed.

zoltanp, 2015-09-20:
I have tried to reproduce the issue on 32 and 64 bit systems, and I coldn't. I made a test circuit with JK-flip-flops used as counters, as on the binary counter I couldn't identify the pins -- I will have to fix that bug, too.

The picture below shows the idea of the test circuit and the output. The "sawtooth" signal on the DAC is regular, so the circuit goes through all of the expected states.

[image]

I can send you the circuit I have made for testing, just please propose a way of transmitting the circuit file. 

The code of the Clock Input component doesn't have any "obvious" issues, but it is too convoluted in my opinion, and it might cause some unexpected behavior. For fully validating the code, I will need more time.
I suspect that there is some random (undefined) behavior in the code, just it needs more investigation.

BenjiWiebe, 2015-09-21:
@zoltanp I think it is maxing out the CPU. But not 'ktechlab', it is X.org that is maxing out the CPU while ktechlab is running. You can email me that circuit at benjiwiebe14 at gmail dot com.

[Here](http://www.benjiwiebe.com/example.circuit) is a circuit that when started, the osc display is smooth at first, but then every couple of seconds will change speed (the signal displayed is not incorrect; the display speed will suddenly slow down and speed up.)

zoltanp, 2015-09-30:
@BenjiWiebe It made me curios: what graphics hardware and driver do you use when the screen update starts to behave strangely? Also: you can change the Display Refresh Rate in settings, on the General tab. Maybe settings a lower or lowest refresh rate (25 or 10 FPS), could work around your issue.

BenjiWiebe, 2015-09-30:
@zoltanp I was not aware of the Display Refresh Rate settings, thanks for pointing that out. I think I have came to the conclusion that the issue was due to the CPU being maxed out. I can not reproduce the issue now. Your circuit worked fine, even when I set the display refresh rate to 100fps. Closing.

P.S. The graphics hardware and driver: Whatever open-source driver comes with Fedora 21 for the driver, and Radeon HD 6970.

closed 2015-09-30 by BenjiWiebe

zoltanp, 2015-10-01:
@BenjiWiebe Good to hear that it works now. Thanks for testing and if you have any issue, feel free to signal it.