ktechlab/ktechlab #30: Why is the components missing?

moar55, 2015-12-17:
[image]

I don't have electronic components option in the plugins , I have been trying to install ktechlab a couple of times now ,and this is the only time it seemed to launch but it is missing many stuff . What should I do?

zoltanp, 2015-12-18:
Hi moar55,

how did you compile and launch KTechLab? With the bundled shell scripts (`simple-setup.sh` and `simple-launch.sh`), or you have used CMake directly? Without the shell scripts, there are quite big chances that the ktechlab plugins are not detected when ktechlab is launched.

Please also note that my current focus is to get an oldel version working (instead of implementing all features of the old version into the new version). That version can be foud here:
https://github.com/ktechlab/ktechlab-0.3/tree/port-0.3.8-kde4-v1
It has more useful functionality than this KDevPlatform based version.

moar55, 2015-12-19:
I have used cMake directly.

zoltanp, 2015-12-26:
Theoretically there is a `cmake uninstall` command. I have never used it, though.
Depending on where/how you installed the other version (in your home directory, or under /opt), maybe you can just delete it.
In the worst case, if the uninstall process (partially) deletes the working version, you can just reinstall the working one.