ktechlab/ktechlab #31: Build error because of type change in allPluginsForExtension

ckorn, 2016-04-16:
This line fails to compile because the parameters of `allPluginsForExtension` have changed:
https://github.com/ktechlab/ktechlab/blob/master/src/interfaces/component/icomponentplugin.cpp#L58

http://api.kde.org/extragear-api/kdevelop-apidocs/kdevplatform/shell/html/classKDevelop_1_1PluginController.html#aab8aa37f1b0fe45a03ceeaa2f88a32fe

Now it requires a `QVariantMap` as type for constraints.

zoltanp, 2016-04-18:
Indeed. I will post a fix soon.

zoltanp, 2016-04-28:
I have pushed a new branch that should fix the build problem:

https://github.com/ktechlab/ktechlab/tree/fix-compile-kdevplatform-1.7

Please let me know if this fixes the buid for you. If yes, I will merge it.