ktechlab/ktechlab #32: Build error related to psimprocessor.cpp

slemonide, 2016-06-27:
When I try to build ktechlab, I get the following error:
```
.../ktechlab/src/lib/simulator/electronics/gpsimprocessor.cpp:769:56: error: 'class pic_processor' has no member named 'W'
  RegisterInfo * info = new RegisterInfo( picProcessor->W );
                                                        ^
```

zoltanp, 2016-06-28:
Hi,
I will try to reproduce and fix this issue. For this could you pleas specify
* which version (branch) of ktechlab do you try to build? master or port-0.3.8-kde4-v1 ? 
* Which version of gpsim library do you have installed (if any)? 
* On which version of which linux distribution?

lfser, 2016-08-26:
Hi 
I have same problem.
Even i cloned "fix-compile-kdevplatform-1.7"
And did git checkout port-0.3.8-kde4-v1

i have this problem : 

> -- Configuring incomplete, errors occurred!
> See also "/home/el1/ktechlab/build-simple/CMakeFiles/CMakeOutput.log".
> See also "/home/el1/ktechlab/build-simple/CMakeFiles/CMakeError.log".
> + log_cmd make -C /home/el1/ktechlab/build-simple/src/core
> + make -C /home/el1/ktechlab/build-simple/src/core
> + tee -a /home/el1/ktechlab/build-simple.log
> make: *** /home/el1/ktechlab/build-simple/src/core: No such file or directory.  Stop.
> + log_cmd make install -j2
> + make install -j2
> + tee -a /home/el1/ktechlab/build-simple.log
> make: *** No rule to make target 'install'.  Stop.

I am on Arch linux 
Kdelibs , Qt4 and gpsim were installed

zoltanp, 2016-09-05:
@lfser The error looks like cmake has become confused after you have checked out the port-0.3.8-kde4-v1 branch. Plese delete the `/home/el1/ktechlab/build-simple/` directory, and then re-run the script. Then it should work (or generate different errors...).

Claws81, 2017-09-22:
I've tried enabling gpsim just now, and build-simple.sh fails with:

```
/home/claus/Programs/ktechlab/src/electronics/gpsimprocessor.cpp: In constructor ‘RegisterSet::RegisterSet(pic_processor*)’:
/home/claus/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:753:56: error: ‘class pic_processor’ has no member named ‘W’
  RegisterInfo * info = new RegisterInfo( picProcessor->W );
                                                        ^
src/electronics/CMakeFiles/electronics.dir/build.make:134: recipe for target 'src/electronics/CMakeFiles/electronics.dir/gpsimprocessor.cpp.o' failed
make[2]: *** [src/electronics/CMakeFiles/electronics.dir/gpsimprocessor.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:849: recipe for target 'src/electronics/CMakeFiles/electronics.dir/all' failed
make[1]: *** [src/electronics/CMakeFiles/electronics.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

CMakeError.log states the following about GPsim (If relevant?):

```
Performing C++ SOURCE FILE Test HAVE_GPSIM_0_26 failed with the following output:
Change Dir: /home/claus/Programs/ktechlab/build-simple/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/gmake" "cmTC_7e1bc/fast"
/usr/bin/gmake -f CMakeFiles/cmTC_7e1bc.dir/build.make CMakeFiles/cmTC_7e1bc.dir/build
gmake[1]: Entering directory '/home/claus/Programs/ktechlab/build-simple/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_7e1bc.dir/src.cxx.o
/usr/bin/c++    -I/usr/include/gpsim -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -Wnon-virtual-dtor -Wno-long-long -Wundef -Wcast-align -Wchar-subscripts -Wall -W -Wpointer-arith -Wformat-security -fno-exceptions -DQT_NO_EXCEPTIONS -fno-check-new -fno-common -Woverloaded-virtual -fno-threadsafe-statics -fvisibility=hidden -Werror=return-type -fvisibility-inlines-hidden -DHAVE_GPSIM_0_26   -fexceptions -UQT_NO_EXCEPTIONS -o CMakeFiles/cmTC_7e1bc.dir/src.cxx.o -c /home/claus/Programs/ktechlab/build-simple/CMakeFiles/CMakeTmp/src.cxx
Linking CXX executable cmTC_7e1bc
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_7e1bc.dir/link.txt --verbose=1
/usr/bin/c++    -Wnon-virtual-dtor -Wno-long-long -Wundef -Wcast-align -Wchar-subscripts -Wall -W -Wpointer-arith -Wformat-security -fno-exceptions -DQT_NO_EXCEPTIONS -fno-check-new -fno-common -Woverloaded-virtual -fno-threadsafe-statics -fvisibility=hidden -Werror=return-type -fvisibility-inlines-hidden -DHAVE_GPSIM_0_26    CMakeFiles/cmTC_7e1bc.dir/src.cxx.o  -o cmTC_7e1bc -rdynamic /usr/lib64/libgpsim.so /usr/lib64/libglib-2.0.so 
/usr/lib64/libgpsim.so: undefined reference to `dlopen'
/usr/lib64/libgpsim.so: undefined reference to `dlclose'
/usr/lib64/libgpsim.so: undefined reference to `dlerror'
/usr/lib64/libgpsim.so: undefined reference to `dlsym'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_7e1bc.dir/build.make:99: recipe for target 'cmTC_7e1bc' failed
gmake[1]: *** [cmTC_7e1bc] Error 1
gmake[1]: Leaving directory '/home/claus/Programs/ktechlab/build-simple/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_7e1bc/fast' failed
gmake: *** [cmTC_7e1bc/fast] Error 2

Source file was:
#include <gpsim/pic-processor.h>
int main() { pic_processor *proc = NULL;
return sizeof (proc->Wreg);
}
```

I pulled the latest master branch which compiles fine, if I don't have gpsim installed.
gpsim on OpenSUSE Leap 42.3 is version 3

closed 2017-09-22 by zoltanp

zoltanp, 2017-09-22:
The logs are relevant from CMakeError.log. Pushed a commit which should fix this issue. 
Please let me know if the latest code from git works. If it does not compile, please reopen this issue or open a new one.

Claws81, 2017-09-24:
Confirmed it now builds.
I tried loading the blink example into the virtual PIC, but with the following message 'Could not find PIC with ID "". '
Don't know what that is about.
I don't own a PIC, just confirming ktechlab compiles successfully. :)

zoltanp, 2017-09-24:
Good to hear that it works :-)
For loading the example in a PIC, you need to create a project first, or
else you get that error message.
Please see tutorial here:

https://zoltanp.github.io/2017/09/ktechlab-pic-prog-sim.html

Foadsf, 2019-02-10:
I also have the same issue as @Claws81 . I'm following the tutorial but got stuck first with the error I have explained [here](http://disq.us/p/1zmuvvf) and then the same message as above.

zoltanp, 2019-02-11:
@Foadsf From the discussion above:

> now when I try to load the pic_blink.c file it says:
> Sorry-KTechLab
> libkatepart not available for constructing editor

You get that error most probably because Kate is not installed. As text editor, KTechlab uses kate's embeddable editor. This is a runtime dependency, not a compile-time one.

Foadsf, 2019-02-11:
@zoltanp Thanks for the immediate response. I am trying the KTechLab on Manjaro-KDe specidially because of these dependencies. I will install Kate  and will report here if it is fixed now.

Foadsf, 2019-02-11:
I installed Kate by `sudo pacman -S Kate` but I still get the error:

[image]

zoltanp, 2019-02-12:
Hmm... On Ubuntu there is a package `katepart` and also `ktexteditor-katepart`. Some of these should fix this runtime error. If still doesn't work after experimentation, I will try to figure out the exact dependency (I have most of KDE desktop installed, so I'm not sure what package is needed).