ktechlab/ktechlab #33: Build Error

dodo158, 2016-07-03:
Hi,  

when I build ktechlab:
clonned from                   https://github.com/ktechlab/ktechlab

linux is new installed for this project: 
Distributor ID:	Ubuntu  i386
Description:	Ubuntu 16.04 LTS
Release:	16.04
Codename:	xenial

It is compiled in KDevelop 4.7.3. thru Your manual, from github. 

error:
/home/dodo/ktechlab/ktechlab2/src/interfaces/component/icomponentplugin.cpp:58:147: error: no matching function for call to ‘KDevelop::IPluginController::allPluginsForExtension(const char [23], QStringList&)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:57:5: error: stray ‘\’ in program

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:57:5: error: missing terminating " character

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:80:78: error: no matching function for call to ‘KDevelop::IProjectFileManager::addFolder(KUrl&, KDevelop::ProjectFolderItem*)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:83:44: error: no matching function for call to ‘KDevelop::IProject::inProject(KUrl&)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:83:44: error: no matching function for call to ‘KDevelop::IProject::inProject(KUrl&)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:94:62: error: no matching function for call to ‘KDevelop::IProjectFileManager::renameFolder(KDevelop::ProjectFolderItem*&, KUrl&)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:97:44: error: no matching function for call to ‘KDevelop::IProject::inProject(KUrl&)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:146:42: error: no matching function for call to ‘KDevelop::IProject::inProject(KUrl&)’

/home/dodo/ktechlab/ktechlab2/tests/ktlproject/ktlprojecttest.cpp:165:43: error: no matching function for call to ‘KDevelop::IProject::inProject(KUrl&)’

Please Help Me Thank You Jozef

zoltanp, 2016-07-05:
Hi,

this issue seems to be duplicate of issue #31 . Compiling the branch fix-compile-kdevplatform-1.7 should fix it. Please test, and it if fixes the compilation, I will merge the branch:

https://github.com/ktechlab/ktechlab/tree/fix-compile-kdevplatform-1.7

dodo158, 2016-07-06:
Hi 

i tested it:

Maybe configurator selected wrong compiler, see this warning:
xtended initializer lists only available with -std=c++11 
I have c++11 installed. 

In file included from /home/dodo/ktechlab/ktechlab2/src/interfaces/component/icomponentplugin.cpp:15:0:
/usr/include/kdevplatform/interfaces/iplugincontroller.h:126:109: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
     Q_SCRIPTABLE virtual IPlugin *pluginForExtension(const QString &extension, const QString& pluginName = {}, const QVariantMap& constraints = {} ) = 0;
                                                                                                             ^
/usr/include/kdevplatform/interfaces/iplugincontroller.h:126:109: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
/usr/include/kdevplatform/interfaces/iplugincontroller.h:126:146: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
     Q_SCRIPTABLE virtual IPlugin *pluginForExtension(const QString &extension, const QString& pluginName = {}, const QVariantMap& constraints = {} ) = 0;
                                                                                                                                                  ^
/usr/include/kdevplatform/interfaces/iplugincontroller.h:126:146: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
/usr/include/kdevplatform/interfaces/iplugincontroller.h:137:112: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
     virtual QList<IPlugin*> allPluginsForExtension(const QString &extension, const QVariantMap& constraints = {}) = 0;
                                                                                                                ^
/usr/include/kdevplatform/interfaces/iplugincontroller.h:137:112: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
/usr/include/kdevplatform/interfaces/iplugincontroller.h:174:113: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
     virtual KPluginInfo::List queryExtensionPlugins(const QString &extension, const QVariantMap& constraints = {}) const = 0;
                                                                                                                 ^
/usr/include/kdevplatform/interfaces/iplugincontroller.h:174:113: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
/home/dodo/ktechlab/ktechlab2/src/interfaces/component/icomponentplugin.cpp: In member function ‘KTechLab::IDocumentPlugin* KTechLab::IComponentPlugin::documentPlugin() const’:
/home/dodo/ktechlab/ktechlab2/src/interfaces/component/icomponentplugin.cpp:58:147: error: no matching function for call to ‘KDevelop::IPluginController::allPluginsForExtension(const char [23], QStringList&)’
     QList<KDevelop::IPlugin*> plugins = KDevelop::Core::self()->pluginController()->allPluginsForExtension( "org.kdevelop.IDocument", constraints );
                                                                                                                                                   ^
In file included from /home/dodo/ktechlab/ktechlab2/src/interfaces/component/icomponentplugin.cpp:15:0:
/usr/include/kdevplatform/interfaces/iplugincontroller.h:137:29: note: candidate: virtual QList<KDevelop::IPlugin*> KDevelop::IPluginController::allPluginsForExtension(const QString&, const QVariantMap&)
     virtual QList<IPlugin*> allPluginsForExtension(const QString &extension, const QVariantMap& constraints = {}) = 0;
                             ^
/usr/include/kdevplatform/interfaces/iplugincontroller.h:137:29: note:   no known conversion for argument 2 from ‘QStringList’ to ‘const QVariantMap& {aka const QMap<QString, QVariant>&}’
src/interfaces/CMakeFiles/ktlinterfaces.dir/build.make:302: recipe for target 'src/interfaces/CMakeFiles/ktlinterfaces.dir/component/icomponentplugin.cpp.o' failed
CMakeFiles/Makefile2:1319: recipe for target 'src/interfaces/CMakeFiles/ktlinterfaces.dir/all' failed
make[2]: *** [src/interfaces/CMakeFxtended initializer lists only available with -std=c++11 iles/ktlinterfaces.dir/component/icomponentplugin.cpp.o] Error 1
xtended initializer lists only available with -std=c++11 
Thank You Jozef

zoltanp, 2016-07-07:
This is strange. It looks like as if kdevplatform-dev has not been incorrectly detected by the buildsystem. Could you please verify what does the build script output about KDEV_PLUGIN_VERSION ? Maybe the simplest would be to upload the output of simple-build.sh somewhere.

I have an ubuntu 16.04 in a virtual machine, and ktechlab compiles in it. I have kdevplatform-dev installed in it. The relevant output I get is:

		-- KDEV_PLUGIN_VERSION (1): 
		-- KDEV_PLUGIN_VERSION (2): #define KDEVELOP_PLUGIN_VERSION 17
		-- KDEV_PLUGIN_VERSION (2) set: 17

As I see, currently building with gpsim support is broken, so for building, you have to uninstall gpsim-dev: `sudo apt-get remove gpsim-dev`. This is different issue than the one you hit whebn you try to compile.

The warnings about c++11 are not the cause of failure. They are generated because kdevplatform build process expects c++11 standard, and ktechlab builds with c++03 (the older one), and the header files from kdevplatform use features not present in the c++03 standard. In this case the presence/absence of those features should not make a difference.

zoltanp, 2016-07-07:
In my virtual machine I have:

		$ apt-cache show kdevplatform-dev
		Package: kdevplatform-dev
		Architecture: i386
		Version: 1.7.3-0ubuntu1
		....

dodo158, 2016-07-11:
In virtual box I have new ubuntu : ubuntu-16.04-desktop-i386.iso 

Next I did this:

sudo apt-get install git kdelib5-dev kdevplatform-dev qt4-dev-tools cmake libglib2.0-dev libeigen2-dev

git clone https://github.com/ktechlab/ktechlab.git

        and I have not gpsim-dev

after start compilation:   sh simple-setup.sh 

 I checked:    -- KDEV_PLUGIN_VERSION and next is correct 

compilation have same errors

Next I continue install: 
I install sudo apt-get install kdevelop-dev

Same errors

zoltanp, 2016-07-26:
Finally I had some time to look into this. In a virtual machine with ubuntu-16.04-32bit, I have managed to _compile_ the 0.4 series ktechlab as following:
```
sudo apt-get install git kdelib5-dev kdevplatform-dev qt4-dev-tools cmake libglib2.0-dev libeigen2-dev
git clone https://github.com/ktechlab/ktechlab.git

cd ktechlab
git checkout fix-compile-kdevplatform-1.7
sudo apt-get remove  libeigen2-dev
sh simple-setup.sh 
```
Unfortunately it does not run, because it fails to create a text editor (that is strange...).

zoltanp, 2016-07-26:
I have good news: the port of 0.3 series, with more working functionality than 0.4 series, builds and runs correctly. I have used the following commands, in the sam e virtual machine:
```
git clone https://github.com/ktechlab/ktechlab.git
cd ktechlab
git checkout port-0.3.8-kde4-v1
sh build-simple.sh
sh run-simple.sh
```

dodo158, 2016-07-26:
sudo apt-get install git kdelibs5-dev kdevplatform-dev qt4-dev-tools cmake libglib2.0-dev libeigen2-dev
git clone https://github.com/ktechlab/ktechlab.git
cd ktechlab
git checkout fix-compile-kdevplatform-1.7
sudo apt-get remove  libeigen2-dev
sh simple-setup.sh   
git checkout port-0.3.8-kde4-v1
sh build-simple.sh
sh run-simple.sh

after this compilation have only some minor warning, and it compiled and program good work. 

Great job thank You.

zoltanp, 2016-07-27:
Nice, have fun with KTechLab :-)

I have a small remark about the commands you have used: some of them are redundant. The following alone should be enough:
```
sudo apt-get install git kdelibs5-dev kdevplatform-dev qt4-dev-tools cmake libglib2.0-dev
git clone https://github.com/ktechlab/ktechlab.git
cd ktechlab
git checkout port-0.3.8-kde4-v1
sh build-simple.sh
sh run-simple.sh
```
`git checkout` switches between branches of a git repository, so anything built between two runs of it should not be needed (otherwise is a bug).
installing and then removing a packet also should not be needed, and if it is necessary, then it is a bug.

I have opened separate issue items for the things that I consider than need to be fixed: #35 #36 #37 . When I will have time, I will fix those.