ktechlab/ktechlab #38: News

alexandre-mbm, 2017-01-26:
Hello! What is this repository? I would want to use Ktechlab on Arch Linux but I only found a package file from 2011. At SourceForge the things are from 2013. And here? Do you a forker?

zoltanp, 2017-01-26:
Hi! Nice to hear that you are interested in KTechLab.

The same people work on this repository as at sourceforge, so this is not a fork, but the "official" KTechLab project. Unfortunately the last release has been quite some time ago.

The current status is that KTechLab recently joined KDE, so development will continue there. This repository will be mostly a mirror of the one at KDE:

https://cgit.kde.org/ktechlab.git/

We are working on creating a new release. Currently the code should compile and run on recent linux distributions, but note that it depends on KDE4 libraries. Most functionality should be working, but there are still several things to be fixed.

Discussions about development will be slowly migrating to the mailinglist at KDE:

https://mail.kde.org/mailman/listinfo/ktechlab-devel

the older mailing list is still in use, at sourceforge:

https://sourceforge.net/p/ktechlab/mailman/ktechlab-devel/

This is the current "website" of KTechLab, but it will be migrated at KDE, at some point:

https://github.com/ktechlab/ktechlab/wiki

If you have any questions, feel free to write here, or to the mailing lists, or at IRC on #ktechlab at freenode.

Have fun!

alexandre-mbm, 2017-01-26:
I want suggest maintain upstream's artifacts outside KDE, here at GitHub.

zoltanp, 2017-01-26:
What do you mean by artifacts? 
I'm planning to make available the source code of any release at GitHub, too, in this repository. Currently only a "git master" is available, and some tags from the old releases.

alexandre-mbm, 2017-01-26:
All source code more all documentation. It is more easy.

zoltanp, 2017-01-26:
The story is that the project has joined KDE to be part of a bigger community, and to hopefully attact more contributors. Curretnly there are a lot of things to be done, and the time of myself and other developers is quite limited. The policy at KDE is that KDE projects have to use KDE's infrastructure. So this github repository will be mostly a mirror of the source code.

Personally I don't observe any significant difference between using Github or KDE's git and bugtracker.

zoltanp, 2017-01-26:
And there is also an official mirror of KDE's git at github, in case you prefer github. KTechLab is also available there:

https://github.com/KDE/ktechlab

alexandre-mbm, 2017-01-26:
https://community.kde.org/Incubator/Projects/KTechLab

alexandre-mbm, 2017-01-26:
> Personally I don't observe any significant difference between using Github or KDE's git and bugtracker.

For example?

zoltanp, 2017-01-26:
I wanted to say that there is no difference. At least for me.

alexandre-mbm, 2017-01-26:
The GitHub system is more visible, independent and engaging.

zoltanp, 2017-01-29:
Fair enough. 
Being independent also means that someone has to do most things himself/herself, and in the case of this project, things have been progressing too slowly to be useful in the foreseeable future.

alexandre-mbm, 2017-01-29:
No, GitHub is a site like there also is.