ktechlab/ktechlab #39: Generate circuit from file

Cassiano-Baltazar, 2017-09-21:
Hi..
Three questions ..
First, does the program generate logic circuits?
second, does it generate a circuit from a file?
Third, Does it work by command line?

zoltanp, 2017-09-22:
Hi,

I'm not sure if I correctly understand the questions.

First question:  You mean something like generating from Verilog or VHDL logic circuits, and programming that into FPGAs? That kind of functionality is currently not implmented.

Second question: What kind of circuit generation you mean? And what kind of file? This application cand save/load its own format of circuit files, and can export images. For PCB design there are other applications available.

Third question: KTechLab is essentially a graphical application; at KDE3 times it could be controlled through a programmatic interface (DCOP), but currently the modern replacement of that (DBUS interface) is not yet implemented.

Cassiano-Baltazar, 2017-09-22:
[sample.txt](https://github.com/ktechlab/ktechlab/files/1324499/sample.txt)
I will explain.
I have a software that receives a Verilog file and merge it and generates an EQN file, following an attached example. Now I need to generate the generated circuit design.

zoltanp, 2017-09-22:
I mostly understand now. I still have the question: how will you use the circuit design? Create a PCB? Or you want to simulate it?

As I see it, KTechLab could be useful for displaying and eventually simulating the content of the EQN file, but for integration you will need to create a converter from EQN format to KTechLab's .circuit file format (which is an XML, for examples see the `examples` directory in the source tree)

Cassiano-Baltazar, 2017-09-22:
I just create circuit desing, like sample in attach.
[image]

zoltanp, 2017-09-22:
It should be possible to display the circuit, but you will need to create a converter from the EQN format to KTechLab's .circuit file format.

Cassiano-Baltazar, 2017-09-22:
That its not a problem..
Can send me a template?

zoltanp, 2017-09-24:
For templates I recommend to create circuits in KTechLab, save them as
.circuit files, and then open the .circuit file with a text viewer/editor.
The .circuit files are XML.
There are some samples in the `examples` directory in the source code, for
example this one :
https://github.com/KDE/ktechlab/blob/master/examples/basic/2-way-switch.circuit

The exact format of the .circuit files is not documented formally; the
reference is the source code inside KTechLab.

Cassiano-Baltazar, 2017-09-24:
ok..
E tem alguma maneira de eu passar o caminho do arquivo por linha de comando e o programa me gerar o jpg/pdf do designer do circuito?

zoltanp, 2017-09-24:
I assume your last question can be translated to :

"And is there any way I can pass the file path by command line and the
program generate me the jpg / pdf from the circuit designer?"

Currently in KTechLab this is not possible programmatically.

I wonder, maybe it would be more simple for you to convert the EQN files to
`dot` files and convert those to images:

https://en.wikipedia.org/wiki/DOT_(graph_description_language)

With a few custom shapes for the logic gates, it would be less effort than
creating .circuit files:

http://www.graphviz.org/wiki/how-create-custom-shapes

With this approach you won't be able to simulate the circuit described by
the EQN files...

Cassiano-Baltazar, 2017-09-24:
Okay, thanks for helping me.