ktechlab/ktechlab #40: Port to Qt5

Petross404, 2017-11-06:
Will ktechlab be ported to qt5?

zoltanp, 2017-11-07:
Yes. It is a matter of time. It will be Qt5 + KDE Frameworks 5.

First I want to remove the dependency on Qt3Support, and then port to Qt5 /
KF5.

There are already a few TODO items about it:

https://cgit.kde.org/ktechlab.git/tree/TODO?id=34b41aa3c95d05f8320e852a9b2a156a8d43fd3d#n70

Petross404, 2017-11-08:
Excellent news. Around xmas holidays I will have time to help if there is anything I can do.

zoltanp, 2017-11-08:
Help is always welcome. Currently there are several areas for contributing:
- code related: porting away from Qt3Support/Kde3Support, porting to
Qt5/KF5; also see the TODO file in the source tree
- KTechLab has joined KDE, but currently does not have a website at KDE;
migrating/reorganizing the current website at github wiki would be useful
- using it and reporting/tracking bugs
- translations, documentation updating, writing

For keeping up with the news, I recommend to subscribe to the developer
mailing list, preferably to the one at kde.org.

machitgarha, 2021-04-17:
It's been ported since 0.50.0. This issue should be closed.

zoltanp, 2021-04-18:
Indeed. Closing.

closed 2021-04-18 by zoltanp