ktechlab/ktechlab #42: Simulation cannot be restarted

gabre, 2018-03-15:
As I see, simulations cannot be restarted now (from t(0), time=0). There should be a way however to restart a simulation.

zoltanp, 2018-03-16:
I agree that there is not way to restart the simulation. Currently it would have limited usefullness, because there is no way to export the simulation data for post-processing, so the time reference is not that relevant.

As a workaround I recommend adding "reset" functionality to the circuit itself. This "reset" could set all of the initial conditions which have been true at time=0. In testing of "real world" circuits this is the equivalent of turning it off and on, or having a reset button.

If you have a "stuck" simulation, then either the circuit is invalid, or you have found a bug. Please note that the traces/oscilloscope do have a reset button, because the stored traces are limited in size (in order to prevent the program of using way too much memory during a long simulation).

gabre, 2018-03-16:
> Currently it would have limited usefullness, because there is no way to export the simulation data for post-processing, so the time reference is not that relevant.

If one uses the simulator to learn about electrical circuits, then it is quite useful, however. :)

zoltanp, 2018-04-01:
Added issue report also in KDE's bugzilla : https://bugs.kde.org/show_bug.cgi?id=392609

I do not have a timeline for fixing this issue. Probably when adding unit tests to the simulator, then I will add also "reset" to the the simulator, and then that reset can be exposed to the GUI.

Of course, contributions are always welcome :-)