ktechlab/ktechlab #43: Circuit seemingly results in memory leak

gabre, 2018-03-15:
The following circuit (when simulation is run) results in continuous memory growth - I do not see why memory shouldn't be constant in this case.

`<!DOCTYPE KTechlab>
<document type="circuit">
 <item x="636" y="284" z="7" type="ec/capacitor" id="capacitor" flip="0" angle="0">
  <data type="number" value="0.001" id="Capacitance"/>
 </item>
 <item x="532" y="244" z="0" type="ec/fixed_voltage" id="fixed_voltage" flip="0" angle="0">
  <data type="number" value="5" id="voltage"/>
 </item>
 <item x="540" y="180" z="4" type="ec/fixed_voltage" id="fixed_voltage__69" flip="0" angle="0">
  <data type="number" value="5" id="voltage"/>
 </item>
 <item x="692" y="356" z="1" type="ec/ground" id="ground" flip="0" angle="270"/>
 <item x="676" y="180" z="3" type="ec/led" id="led" flip="0" angle="0">
  <data type="number" value="1e-15" id="I_S"/>
  <data type="number" value="1" id="N"/>
  <data type="number" value="4.7" id="V_B"/>
  <data type="color" value="#f62a2a" id="0-color"/>
 </item>
 <item x="684" y="244" z="2" type="ec/npnbjt" id="npnbjt" flip="0" angle="0">
  <data type="number" value="100" id="B_F"/>
  <data type="number" value="1" id="B_R"/>
  <data type="number" value="1e-16" id="I_S"/>
  <data type="number" value="1" id="N_F"/>
  <data type="number" value="1" id="N_R"/>
 </item>
 <item x="612" y="180" z="5" type="ec/resistor" id="resistor" flip="0" angle="0">
  <data type="number" value="1" id="resistance"/>
 </item>
 <item x="588" y="244" z="6" type="ec/resistor" id="resistor__84" flip="0" angle="0">
  <data type="number" value="10000" id="resistance"/>
 </item>
 <connector route="69,22,70,22,71,22,72,22,73,22," end-node-parent="resistor" manual-route="1" end-node-is-child="1" start-node-is-child="1" end-node-cid="n1" start-node-parent="fixed_voltage__69" id="connector" start-node-cid="p1"/>
 <connector route="76,30,76,31,76,32,76,33,76,34,76,35,77,35," end-node-parent="capacitor" manual-route="1" end-node-is-child="1" start-node-is-child="1" end-node-cid="n1" start-node-parent="resistor__84" id="connector__100" start-node-cid="p1"/>
 <connector route="83,30,82,30,82,31,82,32,82,33,82,34,82,35,81,35," end-node-parent="capacitor" manual-route="1" end-node-is-child="1" start-node-is-child="1" end-node-cid="p1" start-node-parent="npnbjt" id="connector__101" start-node-cid="b"/>
 <connector route="86,28,86,27,87,27,88,27,88,26,88,25,88,24,88,23,88,22,87,22,86,22," end-node-parent="led" manual-route="1" end-node-is-child="1" start-node-is-child="1" end-node-cid="p1" start-node-parent="npnbjt" id="connector__71" start-node-cid="c"/>
 <connector route="86,32,86,33,86,34,86,35,86,36,86,37,86,38,86,39,86,40,86,41,86,42," end-node-parent="ground" manual-route="1" end-node-is-child="1" start-node-is-child="1" end-node-cid="p1" start-node-parent="npnbjt" id="connector__72" start-node-cid="e"/>
 <connector route="79,22,80,22,81,22,82,22," end-node-parent="led" manual-route="1" end-node-is-child="1" start-node-is-child="1" end-node-cid="n1" start-node-parent="resistor" id="connector__80" start-node-cid="p1"/>
 <connector route="68,30,69,30,70,30," end-node-parent="resistor__84" manual-route="0" end-node-is-child="1" start-node-is-child="1" end-node-cid="n1" start-node-parent="fixed_voltage" id="connector__87" start-node-cid="p1"/>
</document>`

zoltanp, 2018-03-16:
After a quick check: me neither. I will look into it, probably next week.

zoltanp, 2018-03-16:
This looks like a rather classical memory leak:

```
==5624== 15,943,006 (2,505,132 direct, 13,437,874 indirect) bytes in 208,761 blocks are definitely lost in loss record 11,591 of 11,591
==5624==    at 0x402C6BC: operator new(unsigned int) (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5624==    by 0x8258D39: ElementSet::doNonLinear(int, double, double) (elementset.cpp:125)
==5624==    by 0x825600E: Circuit::doNonLogic() (circuit.cpp:478)
==5624==    by 0x81C46D7: Simulator::step() (simulator.cpp:113)
==5624==    by 0x58C9D6A: QMetaObject::activate(QObject*, QMetaObject const*, int, void**) (in /usr/lib/i386-linux-gnu/libQtCore.so.4.8.7)
==5624==    by 0x591F821: QTimer::timeout() (in /usr/lib/i386-linux-gnu/libQtCore.so.4.8.7)
==5624==    by 0x58D520E: QTimer::timerEvent(QTimerEvent*) (in /usr/lib/i386-linux-gnu/libQtCore.so.4.8.7)
==5624==    by 0x58CFD8D: QObject::event(QEvent*) (in /usr/lib/i386-linux-gnu/libQtCore.so.4.8.7)
==5624==    by 0x4956649: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/lib/i386-linux-gnu/libQtGui.so.4.8.7)
==5624==    by 0x495D950: QApplication::notify(QObject*, QEvent*) (in /usr/lib/i386-linux-gnu/libQtGui.so.4.8.7)
==5624==    by 0x631F71B: KApplication::notify(QObject*, QEvent*) (in /usr/lib/libkdeui.so.5.14.16)
==5624==    by 0x58B4579: QCoreApplication::notifyInternal(QObject*, QEvent*) (in /usr/lib/i386-linux-gnu/libQtCore.so.4.8.7)
==5624== 
==5624== LEAK SUMMARY:
==5624==    definitely lost: 2,516,684 bytes in 208,927 blocks
==5624==    indirectly lost: 13,479,149 bytes in 211,535 blocks
==5624==      possibly lost: 18,607 bytes in 274 blocks
==5624==    still reachable: 1,238,768 bytes in 20,975 blocks
==5624==                       of which reachable via heuristic:
==5624==                         newarray           : 8,812 bytes in 169 blocks
==5624==         suppressed: 0 bytes in 0 blocks
```

```
void ElementSet::doNonLinear( int maxIterations, double maxErrorV, double maxErrorI )
{
	QuickVector *p_x_prev = new QuickVector(p_x); // <<<<<<<<, elementset.cpp:125
```

zoltanp, 2018-04-01:
Should be fixed in current master branch( and also in the repository at KDE : https://cgit.kde.org/ktechlab.git/ )

Please test and let me know if there is any issue.

zoltanp, 2018-05-06:
Closing issue, considering it fixed. If it is not, then please reopen.

closed 2018-05-06 by zoltanp