ktechlab/ktechlab #47: AppImage for Linux

probonopd, 2019-01-27:
Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter
- No repositories needed. Suitable/optimized for air-gapped (offline) machines

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

Please let me know if you are interested, I'm happy to help making an AppImage for Linux.

If you have questions, AppImage developers are on #AppImage on irc.freenode.net.

zoltanp, 2019-01-28:
Hi,

thank you for the suggesion; I agree that it is useful to allow people to use KTechlab as easily as possible. 
Currently I use my time available for KTechlab to port it towards Qt5 and KF5, so for me at least this packaging activity has lower priority than porting.

In case someone would like to do this packaging, I'm fine with it and if some help is needed, I will try to help.

zoltanp, 2019-02-01:
Hi @megastallman ,
it would be great if you could do the AppImage packaging. I assume that many people would use it.
If you need any help from my side, please let me know.

People are already asking for stand-alone packages, for example see the first comment here:
https://zoltanp.github.io/2019/01/ktechlab-0-40-1.html

megastallman, 2019-02-04:
Ok, yesterday I've managed to build a qt4 prefix. Now I'm distilling patches. It has less problems than qt3 had long ago...

probonopd, 2019-02-04:
Why are you interested in Qt 4 at this point?

zoltanp, 2019-02-04:
Hope I understood the question correctly: KTechLab master currently depends on Qt4, so it is one of its dependencies. Other notable one kdelibs4, and some more of KDE runtime are needed for running properly.

probonopd, 2019-02-05:
> Currently I use my time available for KTechlab to port it towards Qt5 and KF5, so for me at least this packaging activity has lower priority than porting.

Is this in a compile-able state?

zoltanp, 2019-02-06:
Currently I don't have any Qt5-specific code. Just recently I've managed to fix all of the bugs after getting rid of qt3support. In theory moving from Qt4 to Qt5 should be relatively easy.
First I want to replace the dependencies in the buildsystem and then see how things are going.

megastallman, 2019-02-06:
Ok, now I'd stick with QT4 then.

megastallman, 2019-02-17:
Hi everyone!
I've recently built QT4 without webkit and openssl. Hope these components are not needed for Kdelibs-4.14.38 and Ktecklab.
But now I've got problems with Kdelibs. I still cannot figure out what I miss here. Here is my output: https://hastebin.com/siqukavaqo.bash

zoltanp, 2019-02-18:
Based on the log, Kdelibs depends on Qt compiled with SSL support, so openssl is needed. AFAIK Webkit should not be needed.

```
CMake Error at kdecore/network/ConfigureChecks.cmake:22 (message):
  KDE Requires Qt to be built with SSL support
Call Stack (most recent call first):
  kdecore/CMakeLists.txt:23 (include)
```

Otherwise the feature summary looks good to me.

megastallman, 2019-02-25:
Hi @zoltanp !

Thanks much! I've added the required openssl patches.
Now I've got another problem with Kdelibs build: https://pastebin.com/4VghAV7z
Looks like it needs a static libdbusmenu-qt library... If that is a reason, I've got only a dynamic one(*.so) in my prefix. Have you got any ideas? Maybe I can supply some more info?

zoltanp, 2019-02-27:
Based on the error it does not look like as a dynamic/static linking issue, but a "typical" C++ issue when some virtual methods of a class are not implemented, the class becomes abstract, and thus linking errors are generated around its constructor/destructor.
I suspect that some version of `KDBusMenuExporter` 's dependency is not the one expected.
I do not have any more concrete ideas about the source of the error.

Somewhat related: there is a discussion about cross-building KDE apps for windows (using KF5) on kde-devel mailing list. The mailing list archive does not appear to work (for me?), so I'm linking the code repository mentioned in the discussion:

> TLDR: Here are some scripts to build KDE frameworks and okular statically using gcc/musl and cross-building for mingw: https://github.com/jschultz/kde-static Look in the file patch-kde.sh to see the interesting stuff.

https://github.com/jschultz/kde-static

megastallman, 2019-02-27:
OK, Thanks. I'll take a look...

megastallman, 2019-03-15:
OK, I still miss some of the KDE4 libs, but Ktechlab is getting built if I copy those from my host system.

So the prefix is not ready yet. I'll continue poking with Kdelibs build options.

zoltanp, 2019-03-15:
Nice :-) Having something built with manual intervention is a significant milestone in this project.

megastallman, 2019-03-16:
Now I've got a 46MB Appimage file, that, kind of "works for me".
So I consider to test it under Xubuntu_14.04, OpenSUSE Thumbleweed and Centos VMs. None of them will include any QT installation.

probonopd, 2019-03-16:
> Now I've got a 46MB Appimage file

Where can it be downloaded? Happy to run it through my test suite.

megastallman, 2019-03-17:
I see, it lacks phonon at least... So I'm adding it.

megastallman, 2019-03-17:
https://my-files.ru/2leg0f - This tarball contains everything, but it still lacks some dependencies. Its md5sum is 7f16af4f36aa8f773f6eefb3b44c299e

megastallman, 2019-03-21:
My scripts are residing here: https://gitlab.com/megastallman/ktechlab_qt4_appimage
Now my appimage is working on newer *buntu LiveCDs and on Manjaro. It doesn't yet run on Xubuntu 14.04 and OpenSUSE Leap due to GLIBC problems, so I'm rebuilding on Ubuntu 14.04.

probonopd, 2019-03-21:
Applications run on newer, but not older systems than the one that was used for building. 
https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#creating-appimages-that-are-compatible-with-many-systems

zoltanp, 2019-03-24:
> https://my-files.ru/2leg0f - This tarball contains everything, but it still lacks some dependencies. Its md5sum is 7f16af4f36aa8f773f6eefb3b44c299e

I've tried to run it on XUbuntu 16.04, amd64, and got missing library errors; I assume that it is built for newer system (yep, it's probably time to upgrade this Linux installation :-) ):
```
$ ./KTechlab-x86_64.AppImage 
Current directory is: /tmp/.mount_KTechl6fSlu8
ktechlab: error while loading shared libraries: libpng16.so.16: cannot open shared object file: No such file or directory
```
and after installing `libpng16-16`, I've got this:
```
$ ./KTechlab-x86_64.AppImage 
Current directory is: /tmp/.mount_KTechlViWcxo
ktechlab: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
```

> My scripts are residing here: https://gitlab.com/megastallman/ktechlab_qt4_appimage
> Now my appimage is working on newer *buntu LiveCDs and on Manjaro. It doesn't yet run on Xubuntu 14.04 and OpenSUSE Leap due to GLIBC problems, so I'm rebuilding on Ubuntu 14.04.

Thanks for sharing it!

probonopd, 2019-03-24:
Please use a tool like [linuxdeployqt](https://github.com/probonopd/linuxdeployqt). Just running appimagetool will not bundle all required dependencies, and doing this manually would be very cumbersome.

megastallman, 2019-03-26:
Hi @probonopd !

Looks like Linuxdeployqt should be a much easier way to build it, but I've got some problems here.
I've tried both `./linuxdeployqt-6-x86_64.AppImage /tmp/ktechlab-prefix/usr/bin/ktechlab -bundle-non-qt-libs` and `./linuxdeployqt-6-x86_64.AppImage /tmp/ktechlab-prefix/usr/bin/ktechlab` , but still got the same error. Setting `-qmake=/usr/bin/qmake` didn't help either, as long as I've got the single qmake instance on Ubuntu 14.04 and just qt4, as I need for Ktechlab.
`
linuxdeployqt 5 (commit 37631e5), build 631 built on 2019-01-25 22:47:58 UTC
FHS-like mode with PREFIX, fhsPrefix: "/tmp/ktechlab-prefix/usr"
app-binary: "/tmp/ktechlab-prefix/usr/bin/ktechlab"
appDirPath: "/tmp/ktechlab-prefix"
relativeBinPath: "usr/bin/ktechlab"
ERROR: Could not find any translations in  "/usr/share/qt4/translations"  (developer build?)`

The  "/usr/share/qt4/translations" directory contains *.qm files, though...

probonopd, 2019-03-26:
Does it build the AppImage (but without translations) or does not not create the AppImage at all?

megastallman, 2019-03-27:
It doesn't create anything at all.

probonopd, 2019-03-27:
That's not what should happen. I have opened an issue at https://github.com/probonopd/linuxdeployqt/issues/360.

rosset, 2019-04-18:
0.40.1 is also available for Fedora 30+
https://bodhi.fedoraproject.org/updates/ktechlab-0.40.1-2.fc30
https://koji.fedoraproject.org/koji/buildinfo?buildID=1252276

megastallman, 2019-04-28:
Hi @rosset 
The great thing about Fedora, is that they(or you) are not dropping old libs. Long ago KTechlab_0.3 has disappeared from Ubuntu, but it is still fairly easy to build on Fedora. Version .40 is absolutely easy to build everywhere. That's great. But there are some problems with universal packages, see my post below...

megastallman, 2019-04-28:
@probonopd Hi!
I've patched and rebuilt Linuxdeployqt according to https://github.com/probonopd/linuxdeployqt/issues/360. It grabs QT4 binaries fine. But...
The application still depends on KDElibs, basically on a bigger part of KDE4, including kinit and ksycoca4. It just can't connect to the required daemons.
So the LinuxdeployQT method is fine for QT5 and QT4(after patching the template to qt*.qm) apps, but now I need to get a half of KDE4 with me. I've stuck...
Here is the log on a livecd: https://hastebin.com/tojiriteqe.coffeescript

probonopd, 2019-04-29:
Appears that the reason for the crash is

```
 19-04-28 21:39:25,677 (Fatal) !!!!!!! ERROR !!!!!!! - KHTML default stylesheet version mismatch. Aborting. Check your installation. File used was: /home/xubuntu. Expected STYLE_VERSION 1

(Fatal) !!!!!!! ERROR !!!!!!! - KHTML default stylesheet version mismatch. Aborting. Check your installation. File used was: /home/xubuntu. Expected STYLE_VERSION 1
```

So apparently the application is trying to load KHTML, and possibly KHTML or some component needed by it is not bundled or is not being loaded from the correct (relative) path.

Before we spend lots of time to engineer this for Qt 4, will there be a port to Qt 5 anytime soon? It comes with QtWebEngine which is known to be deployable by linuxdeployqt.

megastallman, 2019-04-29:
Nice question, @probonopd !
@zoltanp , what do you think? QT4 is already deprecated, but we should still bundle it if you are not yet ready with QT5.
I've also heard that QT6 has recently branched. It can get into constant rewriting the same application for  every toolkit version.

zoltanp, 2019-05-02:
@megastallman @probonopd  Hopefully this year there will be a Qt5 /KF5 release of KTechLab. As I read more and more KF5 porting documentation, I see more and more work ahead. Qt5 might be mostly source-compatible with Qt4, but KF5 is not really drop-in replacement KDELibs4. Maybe I could just use Qt5 instead of KF5 where possible [ silly smiley here ].

You can see the progress of KF5 port in the `wip-kf5-try1` branch [1]. Currently it just produces some strange compilation errors.

[1]
https://cgit.kde.org/ktechlab.git/tree/?h=wip-kf5-try1

probonopd, 2019-05-02:
Please ping me once it compiles, then I'll be happy to help with the AppImage. Thanks.

rosset, 2019-05-02:
Do you have CI enabled to compile on each commit? Maybe it helps to engage more people to help on migration/code fix? Thanks.

zoltanp, 2019-05-03:
> Please ping me once it compiles, then I'll be happy to help with the AppImage. Thanks.

Okay. Besides compiling, it should also run :-D

Symbian9, 2019-09-11:
Any success on `KTechLab-0.40.1-*.AppImage` (nightly build) packaging?

Eddieghoul, 2019-09-11:
Nope, just an endless list of dependants that keep propagating. 

⁣Get BlueMail for Android ​

megastallman, 2019-10-11:
Hi everyone!

I've just built an AppImage: 	https://gofile.io/?c=f5aZ7n
Its md5sum is 3985260969bc7650836f451ac1f4df48
I did it in Kubuntu 16.04, so now it runs on Kubuntu 16.04 and 18.04, but looks like I've got bigger problems here. When I run on Centos 8 or Xubuntu 16.04, it claims on lack of some KF5 components and crashes due to heap buffer owerflow. @probonopd , @zoltanp , could you please take a look at the logs: https://hastebin.com/utoteyeciw.sql ?

Symbian9, 2019-10-11:
>  it claims on lack of some KF5 components and crashes due to heap buffer overflow.

Is it possible package `ktechlab` without KF5 components (e.g. `ktechlab-qt5`)? In same way as `kdiff3-qt` packaged:

- https://launchpad.net/ubuntu/trusty/+package/kdiff3-qt

zoltanp, 2019-10-11:
Currently it's not possible to compile KTechLab without KF5 dependencies.
Maybe it would be possible to rewrite (a lot) of code to only use Qt
version of stuff, but it's a lot of work, and currently I don't see any
significant benefit from doing it.

app4soft <notifications@github.com> ezt írta (időpont: 2019. okt. 11., P,
13:21):

megastallman, 2019-10-11:
@zoltanp I would better try to pack any single dependency, piece by piece. It looks much more valuable.

probonopd, 2019-10-11:
> Currently it's not possible to compile KTechLab without KF5 dependencies.

KF5 dependencies are no problem for AppImage. They will be bundled as private copies, inside the AppImage. The end user does not need to install KF5 then.

megastallman, 2019-10-12:
Ok, moving on. I've added the platform plugin. Now the application dies when I press the "New File" button. I've got logs with and without strace here.
https://gofile.io/?c=ECgss8

probonopd, 2019-10-13:
Just guessing, could `(WW) void ItemLibrary::loadItemDescriptions() Could not open file " "/tmp/.mount_*/inst-simple/share/ktechlab/contexthelp/help-English" " ` be causing this? Does the file even exist at that location?

probonopd, 2019-10-13:
Could it be that it should be `usr` instead of `inst-simple/`? Can you point me to the AppImage for download?

zoltanp, 2019-10-13:
> Ok, moving on. I've added the platform plugin. Now the application dies when I press the "New File" button. I've got logs with and without strace here.
> https://gofile.io/?c=ECgss8

Sigh... essentially the same location, same crash; here is the code:

https://github.com/ktechlab/ktechlab/blob/master/src/gui/newfiledlg.cpp#L99

The strace log is useful, it shows that KTechLab doesn't find icons for "source", "text-x-csrc", "application-x-circuit", "application-x-flowcode", "application-x-microbe' (it needs some icon for *all* of these to not crash). Some of these files are installed by KTechLab, some should be part of the system.

Commands I've used to look in the strace log:

List of locations where KTechLab searched for a specific icons:
`grep xubuntu-strace.log -e 'application-x-circuit.* = ' | less`

List of locations where KTechLab managed to load the icon (yes, it's empty):
`grep xubuntu-strace.log -e 'application-x-circuit.* = [^-]' | less`

Still, I assume that KTechLab starts, and maybe opening examples work.

zoltanp, 2019-10-13:
> Just guessing, could `(WW) void ItemLibrary::loadItemDescriptions() Could not open file " "/tmp/.mount_*/inst-simple/share/ktechlab/contexthelp/help-English" " ` be causing this? Does the file even exist at that location?

That line appears on my computer, too. It's a bug (artifact from KF5 porting I guess) and item descriptions will be missing, but KTechLab should not crash because of it.

megastallman, 2019-10-13:
Here is the file itself: https://gofile.io/?c=4jKOld 
md5sum: 501bad7bcbecf1ad7c379d3762c8a0a6

zoltanp, 2019-11-12:
> https://github.com/ktechlab/ktechlab/blob/master/src/gui/newfiledlg.cpp#L99
>
> The strace log is useful, it shows that KTechLab doesn't find icons for "source", "text-x-csrc", "application-x-circuit", [...]

Pushed a fix to master which should fix this crash.
I expect missing icons in the New file... dialog, but no crash.

megastallman, 2019-11-16:
Ok @zoltanp , moving further...
I could even run a simulation from an Appimage build(built on Kubuntu 16.04 VM).
Here is a file dialog screenshot: https://ibb.co/ssKTpmb The black theme is on my Kubuntu 18.04 host, the white - Xubuntu 16.04 VM. As you said, we lack of KF5 icons. Can't even use a dialog.
Though, there is a bigger problem. Another crash when I run the oscilloscope or try to wire a diode: https://hastebin.com/ayarefexer.sql 
I think we should enlist all resources we lack of and I should add them to build.
@probonopd what if I try to add the whole /usr/share/icons/ directory? Can we add it and make it visible to the application?
Still I think that we should bundle all resources. Yeah KF5 comes out to be a very complicated set of software.

probonopd, 2019-11-17:
Yes, all resources you want to use need to be bundled inside the AppDir. Also make sure to load resources from paths relative to the main executable, rather than from absolute compile-time paths.

You could use[ `QString QCoreApplication::applicationDirPath()`](http://doc.qt.io/qt-5/qcoreapplication.html#applicationDirPath) to construct a _relative_ path to `../share/...` from there.

machitgarha, 2021-04-17:
What's the state of this one? I think the project is already migrated to Qt5.

edivaldo-amaral, 2021-04-18:
O Ktechlab é muito difícil de instalar. Seria muito bom manter uma
APPImagem dele.

Atenciosamente:
Edivaldo Amaral Gonçalves

zoltanp, 2021-04-18:
> What's the state of this one? I think the project is already migrated to Qt5.

Yes, KTechLab is working with Qt5 / KF5. I haven't tried to create an AppImage.

Perhaps @megastallman , @probonopd  tried to create an AppImage using some current code of KTechLab ?

megastallman, 2021-04-20:
I had tons of problems with KF5, as I recall. Couldn't make the binaries run rootlessly. The biggest problem where I've stopped - was the content of /usr/share/icons dir that can't be dynamic.
Probably I need to reconsider the approach once again. Thanks for interest, guys.

zoltanp, 2022-06-26:
Cross-linking this discussion with KDE bugzilla

https://bugs.kde.org/show_bug.cgi?id=455969