ktechlab/ktechlab #48: AUR package fails due to dependencies

Foadsf, 2019-02-09:
I tried using [the AUR package](https://aur.archlinux.org/packages/ktechlab-git) but it fails due to dependencies `kdelibs` and `automoc4`.

zoltanp, 2019-02-11:
Apparently that package is missing some dependencies, so someone competent should fix it. I've added a comment there about this issue.

For fixing this issue probably an exact message would still be useful.

Foadsf, 2019-02-11:
Thanks for the immediate response, I appreciate your great work. Some points:

- It would be great if you could add the software to as many repositories / package managers as possible. This will multiply your user base. [The Repology search for KTechLab](https://repology.org/metapackage/ktechlab/versions) says that AUR and Fedora already have some formulas. It would be great if you could add it to Chocolatey to reach the Windows users. I have opened [a ticket here](https://github.com/chocolatey/chocolatey-package-requests/issues/544). If you also add it to LinuxBrew/macBrew then you have almost all the major platforms covered.
- I have mentioned KTechLab in [this list of FOSS for EE](https://www.reddit.com/r/electronics/comments/ap6m45/curated_list_of_awesome_free_and_open_source/). Your thoughts and ideas are very welcomed.

ognarb, 2019-03-27:
Apparently were is a new KTechLab package in the AUR: https://aur.archlinux.org/packages/ktechlab/. It works on my machine ;)

https://chaos.social/@uint8_t/101824393593361799

zoltanp, 2019-03-28:
@ognarb  : cool, thanks for the notification. Currently I do not have a mastodon account, so feel free to send my regards to the packager :-)

Based on this report I'm tempted to close this issue.

zoe1337, 2019-03-28:
@zoltanp you're welcome :)

gustavklopp, 2019-07-10:
> @ognarb : cool, thanks for the notification. Currently I do not have a mastodon account, so feel free to send my regards to the packager :-)
> 
> Based on this report I'm tempted to close this issue.

Don't. It still missing the same dependencies (kdelibs) with this AUR repo also.

zoltanp, 2019-07-10:
Please note that KTechLab master has changed its dependencies: currently it depends on KF5 and not on KDELibs4, like it used to depend several months ago. Maybe this is causing dependency issues.

gustavklopp, 2019-07-10:
Thanks for your answer!
I'm trying to build from source then.

There's a KF5 package on Pacman but I need the developer edition apparently. How can I get this?
https://www.archlinux.org/groups/x86_64/kf5/

zoltanp, 2019-07-10:
I **suppose** that on Arch Linux the typical packages include development files, too, so the "normal" packages should be sufficient.
Disclaimer: I have never used Arch Linux or its derivatives.

gustavklopp, 2019-07-10:
There are missing a few of the modules inside KF5:
From `build-simple.log`:
```
-- Found KF5Config: /usr/lib64/cmake/KF5Config/KF5ConfigConfig.cmake (found version "5.59.0") 
-- Found KF5CoreAddons: /usr/lib64/cmake/KF5CoreAddons/KF5CoreAddonsConfig.cmake (found version "5.59.0") 
**-- Could NOT find KF5DocTools (missing: KF5DocTools_DIR)
-- Could NOT find KF5DocTools: found neither KF5DocToolsConfig.cmake nor kf5doctools-config.cmake** 
-- Found KF5IconThemes: /usr/lib64/cmake/KF5IconThemes/KF5IconThemesConfig.cmake (found version "5.59.0") 
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.8.1") 
-- Found KF5I18n: /usr/lib64/cmake/KF5I18n/KF5I18nConfig.cmake (found version "5.59.0") 
**-- Could NOT find KF5KDELibs4Support (missing: KF5KDELibs4Support_DIR)
-- Could NOT find KF5KDELibs4Support: found neither KF5KDELibs4SupportConfig.cmake nor kf5kdelibs4support-config.cmake** 
**-- Could NOT find KF5KHtml (missing: KF5KHtml_DIR)
-- Could NOT find KF5KHtml: found neither KF5KHtmlConfig.cmake nor kf5khtml-config.cmake** 
-- Found KF5KIO: /usr/lib64/cmake/KF5KIO/KF5KIOConfig.cmake (found version "5.59.0") 
**-- Could NOT find KF5Parts (missing: KF5Parts_DIR)
-- Could NOT find KF5Parts: found neither KF5PartsConfig.cmake nor kf5parts-config.cmake 
-- Could NOT find KF5TextEditor (missing: KF5TextEditor_DIR)
-- Could NOT find KF5TextEditor: found neither KF5TextEditorConfig.cmake nor kf5texteditor-config.cmake** 
-- Found KF5TextWidgets: /usr/lib64/cmake/KF5TextWidgets/KF5TextWidgetsConfig.cmake (found version "5.59.0") 
-- Found KF5Solid: /usr/lib64/cmake/KF5Solid/KF5SolidConfig.cmake (found version "5.59.0") 
-- Found KF5WidgetsAddons: /usr/lib64/cmake/KF5WidgetsAddons/KF5WidgetsAddonsConfig.cmake (found version "5.59.0") 
-- Found KF5XmlGui: /usr/lib64/cmake/KF5XmlGui/KF5XmlGuiConfig.cmake (found version "5.59.0") 
CMake Error at /usr/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  **Could NOT find KF5 (missing: DocTools KDELibs4Support KHtml Parts**
  TextEditor) (found version "5.59.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/ECM/find-modules/FindKF5.cmake:110 (find_package_handle_standard_args)
  CMakeLists.txt:32 (find_package)

-- Configuring incomplete, errors occurred!
```

gustavklopp, 2019-07-10:
What is the way to install KF5-dev on a debian machine? If there a .deb file I can use it to install on mine.

gustavklopp, 2019-07-10:
Okay. I've installed it. I was looking for KF5khtml etc... when these libs are named khtml only in pacman...

gustavklopp, 2019-07-10:
Ok, I've installed the missing packages. 

But cmake isn't finding the DIR in question (### KF5DocTools, KF5KDELibs4Support, KF5KHtml, KF5Parts, KF5TextEditor), whereas I see that these dirs are exactly in the same location of all the other modules (`/usr/lib64/cmake`) that cmake is able to find?

[image]

gustavklopp, 2019-07-10:
Ok, it's working. I've forgotten to delete the cash of Cmake...