ktechlab/ktechlab #49: build instructions not clear --> make  No rule to make target

Foadsf, 2019-02-12:
Following the issues I had with [AUR ](https://github.com/ktechlab/ktechlab/issues/48)and [Manjaro-KDE](https://unix.stackexchange.com/questions/499938/manjaro-guest-on-virtualbox-not-able-to-get-the-full-resolution) I have decided to go for Kubuntu and compile KTechLab from source. On Manjaro-KDE I was able to simply clone this repo and then run the `sudo sh build-simple.sh` to make. But now on Kubuntu I get this error:

> + uname -s
> + echo Linux
> Linux
> + readlink -f build-simple.sh
> + dirname /home/foobar/Documents/source/ktechlab/build-simple.sh
> + SCRIPTDIR=/home/foobar/Documents/source/ktechlab
> + LOGFILE=/home/foobar/Documents/source/ktechlab/build-simple.log
> + date
> + log_cmd echo == starting build at di 12 feb 2019 12:30:13 CET == 
> + tee -a /home/foobar/Documents/source/ktechlab/build-simple.log
> + echo == starting build at di 12 feb 2019 12:30:13 CET == 
> == starting build at di 12 feb 2019 12:30:13 CET == 
> + mkdir -p /home/foobar/Documents/source/ktechlab/build-simple/
> + cd /home/foobar/Documents/source/ktechlab/build-simple/
> + [ -f /home/foobar/Documents/source/ktechlab/build-simple/CMakeCache.txt ]
> + echo buildsystem being generated
> buildsystem being generated
> + log_cmd cmake -DCMAKE_INSTALL_PREFIX=/home/foobar/Documents/source/ktechlab/inst-simple/ /home/foobar/Documents/source/ktechlab
> + tee -a /home/foobar/Documents/source/ktechlab/build-simple.log
> + cmake -DCMAKE_INSTALL_PREFIX=/home/foobar/Documents/source/ktechlab/inst-simple/ /home/foobar/Documents/source/ktechlab
> build-simple.sh: 40: build-simple.sh: cmake: not found
> + UI_HEADERS_TO_GENERATE=
>         ./src/gui/ui_contexthelpwidget.h
>         ./src/gui/ui_generaloptionswidget.h
>         ./src/gui/ui_linkeroptionswidget.h
>         ./src/gui/ui_processingoptionswidget.h
>         ./src/gui/ui_programmerwidget.h
>         ./src/gui/ui_gpasmsettingswidget.h
>         ./src/gui/ui_newprojectwidget.h
>         ./src/gui/ui_newfilewidget.h
>         ./src/gui/ui_outputmethodwidget.h
>         ./src/gui/ui_scopescreenwidget.h
>         ./src/gui/ui_createsubprojectwidget.h
>         ./src/gui/ui_asmformattingwidget.h
>         ./src/gui/ui_oscilloscopewidget.h
>         ./src/gui/ui_microsettingswidget.h
>         ./src/gui/ui_newpinmappingwidget.h
>         ./src/gui/ui_logicwidget.h
>         ./src/gui/ui_sdccoptionswidget.h
>         ./src/gui/ui_picprogrammerconfigwidget.h
>         ./src/gui/ui_gplinksettingswidget.h
>         
> + make -f src/gui/CMakeFiles/gui.dir/build.make ./src/gui/ui_contexthelpwidget.h
> make: src/gui/CMakeFiles/gui.dir/build.make: No such file or directory
> make: *** No rule to make target 'src/gui/CMakeFiles/gui.dir/build.make'.  Stop.

I think there are some steps/dependencies missing. I would appreciate if you could help me know what's wrong and how I can fix this.

zoltanp, 2019-02-12:
>     * build-simple.sh: 40: build-simple.sh: cmake: not found

You need to install `cmake`, e.g. `sudo apt install cmake`

And apparently the build-simple.sh script does not handle this case and fails only a little bit later...

Foadsf, 2019-02-12:
Thanks for the response. Now I get this error messages:

> qmake: could not exec '/usr/lib/x86_64-linux-gnu/qt4/bin/qmake': No such file or directory
> CMake Error at /usr/share/cmake-3.12/Modules/FindQt4.cmake:1313 (message):
>   Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x
> Call Stack (most recent call first):
>   CMakeLists.txt:10 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/foad/Documents/source/ktechlab/build-simple/CMakeFiles/CMakeOutput.log".
> + UI_HEADERS_TO_GENERATE=
>         ./src/gui/ui_contexthelpwidget.h
>         ./src/gui/ui_generaloptionswidget.h
>         ./src/gui/ui_linkeroptionswidget.h
>         ./src/gui/ui_processingoptionswidget.h
>         ./src/gui/ui_programmerwidget.h
>         ./src/gui/ui_gpasmsettingswidget.h
>         ./src/gui/ui_newprojectwidget.h
>         ./src/gui/ui_newfilewidget.h
>         ./src/gui/ui_outputmethodwidget.h
>         ./src/gui/ui_scopescreenwidget.h
>         ./src/gui/ui_createsubprojectwidget.h
>         ./src/gui/ui_asmformattingwidget.h
>         ./src/gui/ui_oscilloscopewidget.h
>         ./src/gui/ui_microsettingswidget.h
>         ./src/gui/ui_newpinmappingwidget.h
>         ./src/gui/ui_logicwidget.h
>         ./src/gui/ui_sdccoptionswidget.h
>         ./src/gui/ui_picprogrammerconfigwidget.h
>         ./src/gui/ui_gplinksettingswidget.h
>         
> + make -f src/gui/CMakeFiles/gui.dir/build.make ./src/gui/ui_contexthelpwidget.h
> make: src/gui/CMakeFiles/gui.dir/build.make: No such file or directory
> make: *** No rule to make target 'src/gui/CMakeFiles/gui.dir/build.make'.  Stop.

The dependencies in [the Readme page](https://github.com/ktechlab/ktechlab) are not very clear. Should I also install `Qt4-dev`, `glib-dev`, `gpim-dev`? What is  the `KDE4 libraries -dev` one? I would appreciate if you oculd mention the specific libraries for Ubuntu.

zoltanp, 2019-02-13:
Good point. This should be improved.
I will look into it when I will have time, or freel free to investigate.

As I remember, in Ubuntu (and Debian, and their derivatives):
* KDE4 libraries: `kdelibs5-dev` (these are actually KDE4 libraries, I don't know why they are called like this)
* Qt4-dev: `libqt4-dev` (this is a dependency of `kdelibs5-dev`, should be automatically installed together with the former one;
* glib-dev: `libglib2.0-dev`
* gpsim-dev: `gpsim-dev` (bewer the typo in your initial post)

For running KTechLab you will need some more KDE components (or you get errors like "katepart not available", or the file opening dialogs will not work)...

Foadsf, 2019-02-13:
Thanks for the response. I hope you don't mind me saying but things like this are crucial for common users like me. It would be a great idea if you could free up a couple of hours to write a complete cross-distro compilation instructions. I will happily help you by testing or as much as my limited knowledge allows other technical supports.

You are doing a great job almost single handedly. I will be happy to donate if you had a Patreon. I could advertise your Patreon on Reddit too. You could also think of a viable business model for this project. For example start a start-up, get some funds/grants (If you are in the EU I can help), and provide premium services like issue fixing, feature requests, consultancy, teaching, publication...