ktechlab/ktechlab #51: cant save scheme

Ehsan-Shahnazi, 2019-04-14:
hi
i cant save scheme because when i click on save menu, the program will close Instantly.

Petross404, 2019-04-14:
Is there any log or crash report?

Ehsan-Shahnazi, 2019-04-14:
no
i can access you with anydesk.
 Can you install this app?

Ehsan-Shahnazi, 2019-04-14:
my distro is kde neon - 64bit

Ehsan-Shahnazi, 2019-04-14:
yet this error is shown:

> Could not start process Cannot talk to klauncher: The name org.kde.klauncher was not provided by any .service files.

[http://bayanbox.ir/download/2800455425887746200/ktechErr.png](url)

Ehsan-Shahnazi, 2019-04-14:
Unfortunately, for a month there are no elements like
triac, diac, zener, etc.

zoltanp, 2019-05-02:
> yet this error is shown:
> 
> > Could not start process Cannot talk to klauncher: The name org.kde.klauncher was not provided by any .service files.
> 
> [http://bayanbox.ir/download/2800455425887746200/ktechErr.png](url)

That error means that not all dependencies are properly installed (as of date of writing you need KDE4 libraries and KDE4 runtime for KTechLab)

zoltanp, 2019-05-02:
> Unfortunately, the program lacks elements like
> triac, diac, zener, etc.

Just a remark: Zener can be "made" from a normal diode, just set its reverse breakdown voltage to the value you want

zoltanp, 2019-05-02:
> Is there any log or crash report?

FYI: There is always a log file in `/tmp/`, named as `/tmp/ktechlab-pid-<pid>-log`. It might contain useful information for debugging.

Ehsan-Shahnazi, 2019-05-02:
can you copy element form QUCS to ktechlab?

zoltanp, 2019-05-03:
If it is a completely new element (component), then no -- you need to write C++ code. 
For existing components some parameters can be adjusted to match the parameters from other simulators.