ktechlab/ktechlab #53: build from build-simple.sh shows errors

HassanAlsamahi, 2019-06-15:
I've downloaded the source code and tried:
sh build-simple.sh

but it shows the following error:

sh build-simple.sh 
+ uname -s
+ echo Linux
Linux
+ readlink -f build-simple.sh
+ dirname /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple.sh
+ SCRIPTDIR=/media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab
+ LOGFILE=/media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple.log
+ date
+ log_cmd echo == starting build at Sat Jun 15 14:04:12 EET 2019 == 
+ echo == starting build at Sat Jun 15 14:04:12 EET 2019 == 
+ tee -a /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple.log
== starting build at Sat Jun 15 14:04:12 EET 2019 == 
+ mkdir -p /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple/
+ cd /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple/
+ [ -f /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple/CMakeCache.txt ]
+ echo buildsystem generated, using it
buildsystem generated, using it
+ log_cmd make install -j2
+ make install -j2
+ tee -a /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build-simple.log
make: *** No rule to make target 'install'.  Stop.

PS: I am using Mxlinux it is debian based

Petross404, 2019-06-15:
Is it built but not installed?

HassanAlsamahi, 2019-06-15:
> Is it built but not installed?

not it is not built

Petross404, 2019-06-15:
Strange. It seems it's unable to find install target and this is after make has finished building the previous target. 

Can you execute `run-simple.sh`?

HassanAlsamahi, 2019-06-15:
no this error shows when I try to execute run-simple.sh:
+ uname -s
+ echo Linux
Linux
+ readlink -f run-simple.sh
+ dirname /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/run-simple.sh
+ SCRIPTDIR=/media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab
+ export PATH=/media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/inst-simple/bin:/home/HassanAlsamahi/anaconda3/bin:/home/HassanAlsamahi/anaconda3/condabin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/sbin:/usr/sbin
+ /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/inst-simple/bin/ktechlab
run-simple.sh: 38: run-simple.sh: /media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/inst-simple/bin/ktechlab: not found

Petross404, 2019-06-15:
Can you do this?
```
cd path/to/repo
rm -rv build/
mkdir build && cd build
cmake ..
make -j1
```

and maybe post the output?

HassanAlsamahi, 2019-06-15:
Here is the output of cmake .. : 

-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could not set up the appstream test. appstreamcli is missing.
-- Looking for __GLIBC__
-- Looking for __GLIBC__ - found
-- Performing Test _OFFT_IS_64BIT
-- Performing Test _OFFT_IS_64BIT - Success
-- Performing Test HAVE_DATE_TIME
-- Performing Test HAVE_DATE_TIME - Success
CMake Warning at CMakeLists.txt:30 (find_package):
  By not providing "FindKF5TextEditor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "KF5TextEditor", but CMake did not find one.

  Could not find a package configuration file provided by "KF5TextEditor"
  with any of the following names:

    KF5TextEditorConfig.cmake
    kf5texteditor-config.cmake

  Add the installation prefix of "KF5TextEditor" to CMAKE_PREFIX_PATH or set
  "KF5TextEditor_DIR" to a directory containing one of the above files.  If
  "KF5TextEditor" provides a separate development package or SDK, be sure it
  has been installed.

-- Could NOT find KF5Config (missing: KF5Config_DIR)
-- Could NOT find KF5Config: found neither KF5ConfigConfig.cmake nor kf5config-config.cmake 
-- Could NOT find KF5CoreAddons (missing: KF5CoreAddons_DIR)
-- Could NOT find KF5CoreAddons: found neither KF5CoreAddonsConfig.cmake nor kf5coreaddons-config.cmake 
-- Could NOT find KF5DocTools (missing: KF5DocTools_DIR)
-- Could NOT find KF5DocTools: found neither KF5DocToolsConfig.cmake nor kf5doctools-config.cmake 
-- Could NOT find KF5IconThemes (missing: KF5IconThemes_DIR)
-- Could NOT find KF5IconThemes: found neither KF5IconThemesConfig.cmake nor kf5iconthemes-config.cmake 
-- Could NOT find KF5I18n (missing: KF5I18n_DIR)
-- Could NOT find KF5I18n: found neither KF5I18nConfig.cmake nor kf5i18n-config.cmake 
-- Could NOT find KF5KDELibs4Support (missing: KF5KDELibs4Support_DIR)
-- Could NOT find KF5KDELibs4Support: found neither KF5KDELibs4SupportConfig.cmake nor kf5kdelibs4support-config.cmake 
-- Could NOT find KF5KHtml (missing: KF5KHtml_DIR)
-- Could NOT find KF5KHtml: found neither KF5KHtmlConfig.cmake nor kf5khtml-config.cmake 
-- Could NOT find KF5KIO (missing: KF5KIO_DIR)
-- Could NOT find KF5KIO: found neither KF5KIOConfig.cmake nor kf5kio-config.cmake 
-- Could NOT find KF5Parts (missing: KF5Parts_DIR)
-- Could NOT find KF5Parts: found neither KF5PartsConfig.cmake nor kf5parts-config.cmake 
-- Could NOT find KF5TextEditor (missing: KF5TextEditor_DIR)
-- Could NOT find KF5TextEditor: found neither KF5TextEditorConfig.cmake nor kf5texteditor-config.cmake 
-- Could NOT find KF5TextWidgets (missing: KF5TextWidgets_DIR)
-- Could NOT find KF5TextWidgets: found neither KF5TextWidgetsConfig.cmake nor kf5textwidgets-config.cmake 
-- Could NOT find KF5Solid (missing: KF5Solid_DIR)
-- Could NOT find KF5Solid: found neither KF5SolidConfig.cmake nor kf5solid-config.cmake 
-- Could NOT find KF5WidgetsAddons (missing: KF5WidgetsAddons_DIR)
-- Could NOT find KF5WidgetsAddons: found neither KF5WidgetsAddonsConfig.cmake nor kf5widgetsaddons-config.cmake 
-- Could NOT find KF5XmlGui (missing: KF5XmlGui_DIR)
-- Could NOT find KF5XmlGui: found neither KF5XmlGuiConfig.cmake nor kf5xmlgui-config.cmake 
CMake Error at /usr/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find KF5 (missing: Config CoreAddons DocTools IconThemes I18n
  KDELibs4Support KHtml KIO Parts TextEditor TextWidgets Solid WidgetsAddons
  XmlGui)
Call Stack (most recent call first):
  /usr/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/ECM/find-modules/FindKF5.cmake:110 (find_package_handle_standard_args)
  CMakeLists.txt:32 (find_package)

-- Configuring incomplete, errors occurred!
See also "/media/HassanAlsamahi/DaTa1/Ktechlab/ktechlab/ktechlab/ktechlab/build/CMakeFiles/CMakeOutput.log".

and the output of make -j1: 
make: *** No targets specified and no makefile found.  Stop.

Petross404, 2019-06-15:
I suppose you have installed KF5 and cmake needs arguments to find it. Can you search where your distro (which is it?) has installed KF5?

HassanAlsamahi, 2019-06-15:
If you can tell me please how to check if KF5 is installed and up to date from terminal, and if installed how to tell cmake where to find it?, I am using MX linux

Petross404, 2019-06-15:
If you use kde or you are using kde software then kf5 is installed. ;p `man apt-get` should inform you on how you can search or install packages. I am a gentoo user so I can't help with your distro, sorry for that.

zoltanp, 2019-06-16:
> If you can tell me please how to check if KF5 is installed and up to date from terminal, and if installed how to tell cmake where to find it?, I am using MX linux

You should install the development packages  (typically their name ends with -dev) for KDE Frameworks 5 (kf5).
As I see, MX linux is based on Debian, so it should be similar to the Ubuntu system I have at hand.
I have the following kf5 packages are installed on my system:

```
libkf5archive-dev
libkf5attica-dev
libkf5auth-bin-dev
libkf5auth-dev
libkf5bookmarks-dev
libkf5codecs-dev
libkf5completion-dev
libkf5config-bin-dev
libkf5config-dev
libkf5configwidgets-dev
libkf5coreaddons-dev
libkf5coreaddons-dev-bin
libkf5crash-dev
libkf5dbusaddons-dev
libkf5dnssd-dev
libkf5doctools-dev
libkf5emoticons-dev
libkf5globalaccel-dev
libkf5guiaddons-dev
libkf5i18n-dev
libkf5iconthemes-dev
libkf5idletime-dev
libkf5itemmodels-dev
libkf5itemviews-dev
libkf5jobwidgets-dev
libkf5kcmutils-dev
libkf5kdelibs4support-dev
libkf5khtml-dev
libkf5kio-dev
libkf5kjs-dev
libkf5mediaplayer-dev
libkf5newstuff-dev
libkf5notifications-dev
libkf5notifyconfig-dev
libkf5package-dev
libkf5parts-dev
libkf5plasma-dev
libkf5plotting-dev
libkf5pty-dev
libkf5service-dev
libkf5solid-dev
libkf5sonnet-dev
libkf5sonnet-dev-bin
libkf5style-dev
libkf5su-dev
libkf5texteditor-dev
libkf5textwidgets-dev
libkf5threadweaver-dev
libkf5unitconversion-dev
libkf5wallet-dev
libkf5webkit-dev
libkf5widgetsaddons-dev
libkf5windowsystem-dev
libkf5xmlgui-dev

```

HassanAlsamahi, 2019-06-16:
@zoltanp thanks I've installed almost all of these packages and it passed the cmake process, but it shows another error after make command 

[  0%] Automatic MOC and UIC for target microbe
Generating MOC predefs moc_predefs.h
Generating MOC compilation mocs_compilation.cpp
[  0%] Built target microbe_autogen
Scanning dependencies of target microbe
[  0%] Building CXX object microbe/CMakeFiles/microbe.dir/btreebase.cpp.o
[  0%] Building CXX object microbe/CMakeFiles/microbe.dir/btreenode.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/main.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/traverser.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/expression.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/pic14.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/variable.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/optimizer.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/instruction.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/microbe.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/parser.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/microbe_autogen/mocs_compilation.cpp.o
[  2%] Linking CXX executable microbe
//usr/lib/x86_64-linux-gnu/libSM.so.6: undefined reference to `uuid_generate@UUID_1.0'
//usr/lib/x86_64-linux-gnu/libSM.so.6: undefined reference to `uuid_unparse_lower@UUID_1.0'
collect2: error: ld returned 1 exit status
microbe/CMakeFiles/microbe.dir/build.make:380: recipe for target 'microbe/microbe' failed
make[2]: *** [microbe/microbe] Error 1
CMakeFiles/Makefile2:150: recipe for target 'microbe/CMakeFiles/microbe.dir/all' failed
make[1]: *** [microbe/CMakeFiles/microbe.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

Petross404, 2019-06-16:
`grep -r /lib64 uuid_generate` or `$ nm -Dn -o --defined-only /lib/* /usr/lib64/* 2> /dev/null | grep '\buuid_generate\b'` should let you know which symbol is causing this.

HassanAlsamahi, 2019-06-16:
it outputs nothing

zoltanp, 2019-06-17:
For me it's obvious that the issue is with the symbols `uuid_generate` and `uuid_unparse_lower`. These should be defined in `/lib/x86_64-linux-gnu/libuuid.so.1`, from package `libuuid1`. Its development package should be `uuid-dev`, but I do not have that package installed (Ubuntu 18.04).

```
$ objdump -T /lib/x86_64-linux-gnu/libuuid.so.1 | grep -e uuid_generate -e uuid_unparse
0000000000002420 g    DF .text	0000000000000040  UUID_1.0    uuid_generate_random
0000000000002560 g    DF .text	00000000000000a3  UUID_2.31   uuid_generate_sha1
0000000000002350 g    DF .text	00000000000000cc  UUIDD_PRIVATE __uuid_generate_random
0000000000002920 g    DF .text	0000000000000002  UUID_1.0    uuid_unparse_lower
00000000000024b0 g    DF .text	00000000000000a3  UUID_2.31   uuid_generate_md5
0000000000002900 g    DF .text	000000000000000c  UUID_1.0    uuid_unparse_upper
0000000000002910 g    DF .text	000000000000000c  UUID_1.0    uuid_unparse
0000000000002330 g    DF .text	0000000000000005  UUID_1.0    uuid_generate_time
0000000000001960 g    DF .text	00000000000005b0  UUIDD_PRIVATE __uuid_generate_time
0000000000002340 g    DF .text	0000000000000005  UUID_2.20   uuid_generate_time_safe
0000000000002460 g    DF .text	0000000000000046  UUID_1.0    uuid_generate

```

I guess that libuuid is not linked into microbe (but why?) or something causes these symbols to not to be found by the linker.

Try running make as
```
make VERBOSE=1
```
and paste here the end of the output [1], the one after `[ 2%] Linking CXX executable microbe`.

[1] https://stackoverflow.com/questions/2670121/using-cmake-with-gnu-make-how-can-i-see-the-exact-commands

Petross404, 2019-06-23:
Any progress so far?

abhishekkumaresan, 2019-07-28:
Hi 
     I want to use KTech labs, I am unable to install the tool in ubuntu 16.04 (Stuck with almost the same errors), Is there any way I can install (like from a stable branch)

Thank you

Eddieghoul, 2019-08-03:
I'd be very interested to see how this all pans out.  I have a similar issue in that it says it cannot find several KD5 dependencies, but then it says it found the newest version of KD5 that I just installed.  cmake doesn't complete and neither does build-simple.sh

Petross404, 2019-08-04:
Build logs guys ;)

Eddieghoul, 2019-08-04:
Sorry about that.  By the way, I'm also using Fedora 30

[CMakeOutput.log](https://github.com/ktechlab/ktechlab/files/3465063/CMakeOutput.log)

Claws81, 2020-01-28:
I am having the same problem as mentioned above. I am running out of available kde-lib packs to install :)

```
buildsystem being generated
+ log_cmd cmake -DCMAKE_INSTALL_PREFIX=/home/claus/ktechlab/inst-simple/ /home/claus/ktechlab
+ cmake -DCMAKE_INSTALL_PREFIX=/home/claus/ktechlab/inst-simple/ /home/claus/ktechlab
+ tee -a /home/claus/ktechlab/build-simple.log
-- Could not set up the appstream test. appstreamcli is missing.
-- Could NOT find KF5KHtml (missing: KF5KHtml_DIR)
-- Could NOT find KF5KHtml: found neither KF5KHtmlConfig.cmake nor kf5khtml-config.cmake 
CMake Error at /usr/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find KF5 (missing: KHtml) (found version "5.45.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/ECM/find-modules/FindKF5.cmake:110 (find_package_handle_standard_args)
  CMakeLists.txt:40 (find_package)

```
CMake log attached

[CMakeOutput.log](https://github.com/ktechlab/ktechlab/files/4124571/CMakeOutput.log)

zoltanp, 2020-01-30:
@Claws81 : You should install package like `libkf5khtml-dev`.

See here:

https://packages.ubuntu.com/search?searchon=contents&keywords=KF5KHtmlConfig.cmake&mode=exactfilename&suite=bionic&arch=any

Which distrubtion are you using?

Claws81, 2020-01-31:
> @Claws81 : You should install package like `libkf5khtml-dev`.
> 
> See here:
> 
> https://packages.ubuntu.com/search?searchon=contents&keywords=KF5KHtmlConfig.cmake&mode=exactfilename&suite=bionic&arch=any
> 
> Which distrubtion are you using?

I'm using Opensuse 15.0 Leap, so the file is called khtml-devel-5.45.0-lp150.1.1.x86_64.rpm instead - what the h... That is just stupid.
I just found it, using your reply and it is building the sources right now :)

mateusnazarioc, 2020-03-05:
I'm having trouble to install on Kubuntu 18.04, guys.

Output after running build-simple.sh:

+ uname -s
+ echo Linux
Linux
+ readlink -f build-simple.sh
+ dirname /home/mateus/ktechlab/build-simple.sh
+ SCRIPTDIR=/home/mateus/ktechlab
+ LOGFILE=/home/mateus/ktechlab/build-simple.log
+ date
+ log_cmd echo == starting build at qui mar  5 17:41:54 -03 2020 == 
+ echo == starting build at qui mar  5 17:41:54 -03 2020 == 
+ tee -a /home/mateus/ktechlab/build-simple.log
== starting build at qui mar  5 17:41:54 -03 2020 == 
+ mkdir -p /home/mateus/ktechlab/build-simple/
+ cd /home/mateus/ktechlab/build-simple/
+ [ -f /home/mateus/ktechlab/build-simple/Makefile ]
+ echo buildsystem being generated
buildsystem being generated
+ log_cmd cmake -DCMAKE_INSTALL_PREFIX=/home/mateus/ktechlab/inst-simple/ /home/mateus/ktechlab
+ cmake+  -DCMAKE_INSTALL_PREFIX=/home/mateus/ktechlab/inst-simple/ /home/mateus/ktechlab
tee -a /home/mateus/ktechlab/build-simple.log
CMake Warning at CMakeLists.txt:38 (find_package):
  By not providing "FindKF5TextEditor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "KF5TextEditor", but CMake did not find one.

  Could not find a package configuration file provided by "KF5TextEditor"
  with any of the following names:

    KF5TextEditorConfig.cmake
    kf5texteditor-config.cmake

  Add the installation prefix of "KF5TextEditor" to CMAKE_PREFIX_PATH or set
  "KF5TextEditor_DIR" to a directory containing one of the above files.  If
  "KF5TextEditor" provides a separate development package or SDK, be sure it
  has been installed.

-- Could NOT find KF5DocTools (missing: KF5DocTools_DIR)
-- Could NOT find KF5DocTools: found neither KF5DocToolsConfig.cmake nor kf5doctools-config.cmake 
-- Could NOT find KF5KDELibs4Support (missing: KF5KDELibs4Support_DIR)
-- Could NOT find KF5KDELibs4Support: found neither KF5KDELibs4SupportConfig.cmake nor kf5kdelibs4support-config.cmake 
-- Could NOT find KF5TextEditor (missing: KF5TextEditor_DIR)
-- Could NOT find KF5TextEditor: found neither KF5TextEditorConfig.cmake nor kf5texteditor-config.cmake 
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find KF5 (missing: DocTools KDELibs4Support TextEditor) (found
  version "5.44.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/ECM/find-modules/FindKF5.cmake:110 (find_package_handle_standard_args)
  CMakeLists.txt:40 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/mateus/ktechlab/build-simple/CMakeFiles/CMakeOutput.log".
+ log_cmd make install -j2
+ make install -j2
+ tee -a /home/mateus/ktechlab/build-simple.log
make: *** No rule to make target 'install'.  Stop.

zoltanp, 2020-03-09:
Hi @mateusnazarioc ,

based on the cmake errors, you need to install some development packages on your system:

> -- Could NOT find KF5DocTools (missing: KF5DocTools_DIR)
> -- Could NOT find KF5KDELibs4Support (missing: KF5KDELibs4Support_DIR)
> -- Could NOT find KF5TextEditor (missing: KF5TextEditor_DIR)

These are at least: `libkf5doctools-dev`, `libkf5texteditor-dev`, `libkf5kdelibs4support-dev`. In case you get other errors if missing packages, you can search on them on your system with `apt search <name of the not found component>`.

JihedChaibi, 2021-05-17:
H

> Hi @mateusnazarioc ,
>
> based on the cmake errors, you need to install some development packages on your system:
>
>
> These are at least: `libkf5doctools-dev`, `libkf5texteditor-dev`, [...]

Had the same issue, installing these packages solved it.

omarirfa, 2024-01-14:
> Hi @mateusnazarioc ,
>
> based on the cmake errors, you need to install some development packages on your system:
>
>
> These are at least: `libkf5doctools-dev`, `libkf5texteditor-dev`, [...]

This solved my KF5 issue as well. Thank you!