ktechlab/ktechlab #55: gspimprocessor.cpp compiling error

balebanov, 2019-07-14:
Good evening! 

I've got an error after "cmake" command. build-simple.sh runned successfully. Here is my error:

`
Scanning dependencies of target microbe_autogen
[  0%] Automatic MOC and UIC for target microbe
[  0%] Built target microbe_autogen
Scanning dependencies of target microbe
[  0%] Building CXX object microbe/CMakeFiles/microbe.dir/btreebase.cpp.o
[  0%] Building CXX object microbe/CMakeFiles/microbe.dir/btreenode.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/main.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/traverser.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/expression.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/pic14.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/variable.cpp.o
[  1%] Building CXX object microbe/CMakeFiles/microbe.dir/optimizer.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/instruction.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/microbe.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/parser.cpp.o
[  2%] Building CXX object microbe/CMakeFiles/microbe.dir/microbe_autogen/mocs_compilation.cpp.o
[  2%] Linking CXX executable microbe
[  2%] Built target microbe
Scanning dependencies of target test_ktechlab_autogen
[  2%] Automatic MOC and UIC for target test_ktechlab
AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/canvas.cpp"
The file includes the moc file "canvas.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/canvasitemparts.cpp"
The file includes the moc file "canvasitemparts.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/canvasitems.cpp"
The file includes the moc file "canvasitems.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/circuitview.cpp"
The file includes the moc file "circuitview.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/cnitem.cpp"
The file includes the moc file "cnitem.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/canvasmanipulator.cpp"
The file includes the moc file "canvasmanipulator.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/cnitemgroup.cpp"
The file includes the moc file "cnitemgroup.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/core/diagnosticstyle.cpp"
The file includes the moc file "diagnosticstyle.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

AutoMoc warning
---------------
  "/home/user/Загрузки/Programs/ktechlab/src/connector.cpp"
The file includes the moc file "connector.moc", but does not contain a Q_OBJECT, Q_GADGET, Q_NAMESPACE, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY, K_PLUGIN_CLASS_WITH_JSON, K_PLUGIN_FACTORY_WITH_JSON, K_PLUGIN_FACTORY or K_PLUGIN_CLASS_WITH_JSON macro.

[... 846 lines cut ...]
[ 32%] Building CXX object src/CMakeFiles/test_ktechlab.dir/electronics/gpsimprocessor.cpp.o
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp: In constructor ‘RegisterSet::RegisterSet(pic_processor*)’:
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:761:56: error: ‘class pic_processor’ has no member named ‘W’
  RegisterInfo * info = new RegisterInfo( picProcessor->W );
                                                        ^
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp: In destructor ‘RegisterSet::~RegisterSet()’:
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:772:26: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for ( unsigned i = 0; i < m_registers.size(); ++i )
                        ~~^~~~~~~~~~~~~~~~~~~~
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp: In member function ‘RegisterInfo* RegisterSet::fromAddress(unsigned int)’:
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:779:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  return (address < m_registers.size()) ? m_registers[address] : 0l;
          ~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp: In member function ‘void RegisterSet::update()’:
/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:804:26: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for ( unsigned i = 0; i < m_registers.size(); ++i )
                        ~~^~~~~~~~~~~~~~~~~~~~
src/CMakeFiles/test_ktechlab.dir/build.make:2304: recipe for target 'src/CMakeFiles/test_ktechlab.dir/electronics/gpsimprocessor.cpp.o' failed
make[2]: *** [src/CMakeFiles/test_ktechlab.dir/electronics/gpsimprocessor.cpp.o] Error 1
CMakeFiles/Makefile2:278: recipe for target 'src/CMakeFiles/test_ktechlab.dir/all' failed
make[1]: *** [src/CMakeFiles/test_ktechlab.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

`

Petross404, 2019-07-18:
Can you put inside code tags so it's easier for the eyes?

Thanks.

balebanov, 2019-07-18:
> Can you put inside code tags so it's easier for the eyes?
> 
> Thanks.

Sorry for that, I wonder why inserting code function doesn't work for me. I've put that to PasteBin: https://pastebin.com/iWigvB01
Thanks

Petross404, 2019-07-20:
Re-create the build directory (or erase all it's contents) and try to run and give us the following:

```bash
cd pat/to/build
rm -rv ./*
cmake .. && make -j1
```

So we can see what cmake does (not that this is it's fault) and of course run make with -j1 so it can stop immediately after the very first error. I can't quite address where the error happened in your previous log.

balebanov, 2019-07-21:
https://pastebin.com/7zr3GH7a
Here it is. I skipped pasting there generation of build files, it was successful

balebanov, 2019-07-21:
What does "W" mean? Should I do something with source code files?

`/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:761:56: error: ‘class pic_processor’ has no member named ‘W’
  RegisterInfo * info = new RegisterInfo( picProcessor->W );`

Petross404, 2019-07-22:
Indeed 
```bash
.../Programs/ktechlab/src/electronics/gpsimprocessor.cpp:761:56: 
error: ‘class pic_processor’ has no member named ‘W’
RegisterInfo * info = new RegisterInfo( picProcessor->W );
                                                        ^
```

And 
```cpp
#if defined(HAVE_GPSIM_0_26)
	RegisterInfo * info = new RegisterInfo( picProcessor->Wreg ); // is tihs correct for "W" member? TODO
#else
	RegisterInfo * info = new RegisterInfo( picProcessor->W );
#endif
``` 
[informs](https://github.com/ktechlab/ktechlab/blob/master/src/electronics/gpsimprocessor.cpp#L758) us that one should use GPSIM 0.26. Back at the [CMakeLists.txt](https://github.com/ktechlab/ktechlab/blob/master/CMakeLists.txt#L58) I see that GSPIM is indeed required but I can't see any version required.

Please post your GSPIM version and how you installed it in case it's older than 0.26.

balebanov, 2019-07-22:
@Petross404 I've remove and install again GPSIM 0.31.0 version. I used an instruction in official site of GPSIM: http://gpsim.sourceforge.net/#install
I haven't got any error during its installation. Then I run again in ktechlab directory: 
`cmake .. && make -j1`
And got same error:
`/home/user/Загрузки/Programs/ktechlab/src/electronics/gpsimprocessor.cpp:761:56: error: ‘class pic_processor’ has no member named ‘W’
  RegisterInfo * info = new RegisterInfo( picProcessor->W );`

Petross404, 2019-07-23:
It seems you installed it without a package manager but anyway. Also as a side note, it's best to rm your build directory before retrying after a failed compilation.

You have install version 0.31 and yet `HAVE_GPSIM_0_26` is undefined? Why it doesn't compile 	`RegisterInfo * info = new RegisterInfo( picProcessor->Wreg );` instead?

If you know a little about cmake you could use `message()` command to ouput the value of `HAVE_GPSIM_0_26` to debug the CMakeLists.txt. It's late here, so if you find it difficult ping me tomorrow.

Or you can wait the author to answer as I don't have other ideas. Sorry.

All the best,
Petros

balebanov, 2019-07-23:
I've just noticed this part of output of `cmake ..`:

`-- Performing Test HAVE_GPSIM_0_26`
`-- Performing Test HAVE_GPSIM_0_26 - Failed`
`-- Found GPSim: /usr/local/lib/libgpsim.so  `

So I don't know why this test returned **Failed**. I found GPSIM in my package manager and installed it too. About your question 

> Where is gpsim-0.31 installed?

I can tell you that just dowloaded .tar.gz file from their site, unpacked it, compiled and runned `sudo make install`, so I've found GPSIM files in `/usr/local/include/gpsim` (C++ header files) and `/usr/local/lib` (.so libraries).

Regards.

Petross404, 2019-07-24:
CMake output is often useful despite being succesful ;) What is your distro and what version of GPSIM is the greatest? Also can you uninstall the manually compiled GPSIM?

> I've just noticed this part of output of cmake ..:-- Performing Test HAVE_GPSIM_0_26-- Performing Test HAVE_GPSIM_0_26 - Failed-- Found GPSim: /usr/local/lib/libgpsim.so
--

Let me look where this variable is defined in cmakelists.txt. Perphaps `HAVE_GPSIM_0_26` mean only this, having 0.26 version, not a greater one.

balebanov, 2019-07-24:
Petros, I use LM 19.1. Just uninstalled GPSIM which was compiled: `sudo make uninstall`. I found gpsim-0.26.1 (there's no 0.26.0), should I install it instead of 0.31 which installed by package manager?

adrian-the-git, 2019-12-31:
gpsim renamed `W` to `Wreg` sometime during the 0.26.x releases ([commit 2213](https://sourceforge.net/p/gpsim/code/2213/tree//trunk/ChangeLog)) and the cmake module `FindGPSim.cmake` does the right thing by attempting to compile code referencing `pic_processor->Wreg` and setting `HAVE_GPSIM_0_26` when that compilation succeeds.

The problem arises because a gpsim header called `breakpoints.h` tries to `#include "../config.h"`, which usually fails, preventing `HAVE_GPSIM_0_26` from being set (and presumably, faking this flag would be insufficient because the `../config.h` file is still missing). This bug has shown up [elsewhere](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=939338).

Until gpsim fixes their headers, you can probably fix this with a kludge like:

```
--- a/cmake/modules/FindGPSim.cmake
+++ b/cmake/modules/FindGPSim.cmake
@@ -46,7 +46,12 @@ find_path(GPSim_INCLUDE_DIR
 find_library(GPSim_LIBRARY NAMES gpsim)
 
 if (GPSim_INCLUDE_DIR AND GPSim_LIBRARY AND GLib_FOUND)
-    set(GPSim_INCLUDE_DIRS ${GPSim_INCLUDE_DIR} ${GLib_INCLUDE_DIRS})
+    # Make a fake config.h to satisfy GPSim's broken headers
+    set(GPSim_KLUDGE_DIR "${CMAKE_CURRENT_BINARY_DIR}${CMAKE_FILES_DIRECTORY}/kludge/include")
+    file(MAKE_DIRECTORY "${GPSim_KLUDGE_DIR}")
+    file(TOUCH "${GPSim_KLUDGE_DIR}/../config.h")
+
+    set(GPSim_INCLUDE_DIRS ${GPSim_INCLUDE_DIR} ${GLib_INCLUDE_DIRS} ${GPSim_KLUDGE_DIR})
     set(GPSim_LIBRARIES ${GPSim_LIBRARY} ${GLib_LIBRARY})
```

...although someone with better knowledge of cmake might know of better path variables to use. @balebanov looks like you've moved on, but if you're still interested you can likely build ktechlab with this fix.

zoltanp, 2020-01-03:
@adrian-the-git Thanks for the investigation! If this patch fixes the issue, then I will add it as a commit to the repository.

Note to self:
1. There is another open issue related to FreeBSD, might be the same:
https://github.com/ktechlab/ktechlab/issues/57

2. To notify gpsim developers about (1) broken header issue and (2) discuss about API compatibility, and ask for exposing version information their installed headers

zoltanp, 2020-01-16:
Pushed a fix for this issue into git master. Feel free to test.

Remaining TODO for me (or for a volunteer):

* To notify gpsim developers about (1) broken header issue and (2) discuss about API compatibility, and ask for exposing version information in their installed headers

zoltanp, 2020-01-18:
Broken header issue:
https://sourceforge.net/p/gpsim/bugs/253/

Request for version information:
https://sourceforge.net/p/gpsim/feature-requests/63/

Closing ticket as it is fixed and relevant issues reported upstream. If it is not fixed, feel free to reopen.

closed 2020-01-18 by zoltanp