ktechlab/ktechlab #56: Compilation Error

PerhapsSomeone, 2019-10-09:
Compiling ktechlab with the standard ./build-simple.sh script generates the Makefile successfully, but compiling fails with the following message:
`
/home/snip/Data/ktechlab/src/core/main.cpp: In function ‘int main(int, char**)’:
/home/snip/Data/ktechlab/src/core/main.cpp:43:37: error: ‘KAboutLicense::LicenseKey’ is not a class or namespace
                      KAboutLicense::LicenseKey::GPL_V2,
`

I have installed all needed libraries to get the Makefiles working properly, but this is not working.

Petross404, 2019-10-09:
What if you comment out this line? Have you installed KF5?

zoltanp, 2019-10-10:
I'm aware that the dependencies of KTechLab currently are not well documented. Please note that the current git master code depends on KDE Frameworks 5 and not on KDELibs4.

You will need development packages for KDE Frameworks 5, Qt5 and optionally gpsim development files.
In the future I'm planning to update the readme and have an exact list.

Meanwhile: which operating system / linux distribution are you using? What is the version of KDE 5 and Qt 5 there?

megastallman, 2019-10-10:
Guys, I can build Ktechlab successfully on Kubuntu 18.04 after installing `apt install -y libkf5xmlgui-dev libkf5solid-dev libkf5texteditor-dev libkf5doctools-dev libkf5khtml-dev libkf5kdelibs4support-dev gpsim-dev git cmake build-essential kubuntu-desktop extra-cmake-modules pkg-config libglib2.0-dev`.
When I try building it on Kubuntu 16.04, to comply with AppImage requirements, I'm getting the same error @PerhapsSomeone gets.
That all looks like software versions incompatibilities.

zoltanp, 2019-10-10:
@megastallman  Sounds like either a bug in KTechLab (missing includes, dependencies?) or some incompatibility in KDE (hope not). I will try to reproduce this.

Note to self: [documentation about KAboutLicense](https://api.kde.org/frameworks/kcoreaddons/html/classKAboutLicense.html)

zoltanp, 2019-10-10:
I've pushed the branch `fix-kubuntu-16.04-build` ( https://github.com/ktechlab/ktechlab/tree/fix-kubuntu-16.04-build ) to github. If it fixes this issue (and doesn't break anything else) then let me know and I will merge it to master.

megastallman, 2019-10-11:
Thanks @zoltanp !
It builds and runs on Kubuntu 16.04 now.

@PerhapsSomeone :
Those are K 18.04 dependencies: `apt install -y libkf5xmlgui-dev libkf5solid-dev libkf5texteditor-dev libkf5doctools-dev libkf5khtml-dev libkf5kdelibs4support-dev gpsim-dev git cmake build-essential kubuntu-desktop extra-cmake-modules pkg-config libglib2.0-dev`
And those - are for 16.04: `apt install -y libkf5xmlgui-dev libkf5solid-dev libkf5texteditor-dev  libkf5khtml-dev libkf5kdelibs4support-dev gpsim-dev git cmake build-essential kubuntu-desktop extra-cmake-modules pkg-config libglib2.0-dev`

PerhapsSomeone, 2019-10-11:
@zoltanp That works!
Thanks a lot.

closed 2019-10-11 by PerhapsSomeone

zoltanp, 2019-10-13:
Merged to master.