ktechlab/ktechlab #57: The build fails: no member named 'W' in 'pic_processor'

yurivict, 2019-10-25:
```
/usr/ports/cad/ktechlab/work/ktechlab-0.40.1-63-gc53055fa/src/electronics/gpsimprocessor.cpp:761:56: error: no member named 'W' in 'pic_processor'
        RegisterInfo * info = new RegisterInfo( picProcessor->W );
                                                ~~~~~~~~~~~~  ^
```

0.40.1-63-gc53055fa
gpsim-0.31.0 is installed.
FreeBSD 12

zoltanp, 2019-10-25:
Apparently gpsim is too new for KTechLab. I have only tested with gpsim 0.30.
As I see, the issue is that the buildsystem fails to detect that gpsim is newer than 0.26 and uses the wrong define at:

https://github.com/ktechlab/ktechlab/blob/master/src/electronics/gpsimprocessor.cpp#L758

Will need to tweak this check in buildsystem:

https://github.com/ktechlab/ktechlab/blob/master/cmake/modules/FindGPSim.cmake#L59

For now this is just a diagnosis, but when I will have more time I will look into fixing it.
One workaround could be to downgrade gpsim to 0.30. I'm not sure if that is feasable in your case.

yurivict, 2019-10-25:
> One workaround could be to downgrade gpsim to 0.30. I'm not sure if that is feasable in your case.

I am trying to create a FreeBSD port for ktechlab, and can't downgrade the gpsim package.

You should try to support the new gpsim interface.

zoltanp, 2019-12-25:
Tested with Debian unstable, gpsim 0.31.0 and the gpsim detection has
worked correctly. I assume that the issue is related specifically to
FreeBSD. Will test it with a virtual machine.

zoltanp, 2020-01-16:
Tested in virtual machine with FreeBSD, reproduced (several) issues.
I've pushed a fix which makes GPSim >= 0.26 detection work, but the compilation currently still fails.

@yurivict I see some patches of KTechLab at the FreeBSD port [1] . I'd like to merge/apply the generic patches (which do not break other OSs). Is that OK with you?

[1] https://svnweb.freebsd.org/ports/head/cad/ktechlab/files/

yurivict, 2020-01-16:
> Is that OK with you?

Yes, please do.

zoltanp, 2020-01-18:
I've compiled and ran vanilla ktechlab (master, d45f0cd ) on FreeBSD.

For updating the port, the following changes should be applied:

- in Makefile the below line should not be needed; glibconfig.h should be found automatically:
`CXXFLAGS+=      -I${LOCALBASE}/lib/glib-2.0/include`

- the patch `patch-doc_en_CMakeLists.txt` will need an update, because it will fail to apply; for document location I just copied other KDE project's install location

- the patch `patch-src_electronics_gpsimprocessor.cpp` should not be needed anymore; gpsim >= 0.26 detection should work now correctly

- the patch `patch-src_electronics_port.cpp` has been integrated into the official codebase; it is not needed anymore and it wil fail to apply

If you have issues, please let me know.

yurivict, 2020-01-19:
Thank you for helping with the FreeBSD port.

I just updated it following your recommendations.

closed 2020-01-19 by yurivict