ktechlab/ktechlab #60: Segfault when changing element value

abalfoort, 2020-12-15:
Whenever I add an item to the circuit and want to change the element's value Ktechlab segfaults:

```
sh run-simple.sh
+ case "$(uname -s)" in
++ uname -s
+ echo Linux
Linux
+++ readlink -f run-simple.sh
++ dirname /home/solydxk/Downloads/ktechlab-master/run-simple.sh
+ SCRIPTDIR=/home/solydxk/Downloads/ktechlab-master
+ . /home/solydxk/Downloads/ktechlab-master/build-simple/prefix.sh
++ export PATH=/home/solydxk/Downloads/ktechlab-master/inst-simple/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
++ PATH=/home/solydxk/Downloads/ktechlab-master/inst-simple/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
++ export XDG_DATA_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-simple/share:/usr/share/xfce4:/usr/local/share/:/usr/share/:/usr/share
++ XDG_DATA_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-simple/share:/usr/share/xfce4:/usr/local/share/:/usr/share/:/usr/share
++ export XDG_CONFIG_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-simple/etc/xdg:/etc/xdg
++ XDG_CONFIG_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-simple/etc/xdg:/etc/xdg
++ export QT_PLUGIN_PATH=/home/solydxk/Downloads/ktechlab-master/inst-simple/lib/x86_64-linux-gnu/plugins:
++ QT_PLUGIN_PATH=/home/solydxk/Downloads/ktechlab-master/inst-simple/lib/x86_64-linux-gnu/plugins:
++ export QML2_IMPORT_PATH=/home/solydxk/Downloads/ktechlab-master/inst-simple/lib/x86_64-linux-gnu/qml:
++ QML2_IMPORT_PATH=/home/solydxk/Downloads/ktechlab-master/inst-simple/lib/x86_64-linux-gnu/qml:
+ /home/solydxk/Downloads/ktechlab-master/inst-simple/bin/ktechlab
qt5ct: using qt5ct plugin
kf5.kservice.sycoca: Service type not found: "application/x-flowcode"
kf5.kservice.sycoca: Service type not found: "application/x-microbe"
kf5.kservice.sycoca: Service type not found: "application/x-circuit"
kf5.kservice.sycoca: Service type not found: "application/x-ktechlab"
Shortcut for action  "file_new" "New File" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
qt5ct: D-Bus global menu: no
org.kde.ktechlab: no available sizes for  "C (.c)"
Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/libexec/kf5/klauncher'
kdeinit5: Launched KLauncher, pid = 1705, result = 0
qt5ct: using qt5ct plugin
Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
kdeinit5: opened connection to :0.0
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
run-simple.sh: line 39:  1700 Segmentation fault      "$SCRIPTDIR/inst-simple/bin/ktechlab"
```

gdb doesn't tell me more:
```
gdb -p 1725
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 1725
[New LWP 1726]
[New LWP 1727]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007f3b55310819 in __GI___poll (fds=0x56463e5d2890, nfds=4, timeout=17)
    at ../sysdeps/unix/sysv/linux/poll.c:29
29	../sysdeps/unix/sysv/linux/poll.c: No such file or directory.
(gdb) cont
Continuing.

Thread 1 "ktechlab" received signal SIGSEGV, Segmentation fault.
0x00007f3b566a5631 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
```

zoltanp, 2020-12-15:
Hi,
in case you are building KTechLab from sources, can you try building it with `build-developer.sh` and running it with `run-developer.sh` ? This way the Address Sanitiser feature of GCC is turned on, which should catch any crashes and dump some useful information about it.

abalfoort, 2020-12-15:
Here it is:

```
qt5ct: using qt5ct plugin
kf5.kservice.sycoca: Service type not found: "application/x-ktechlab"
kf5.kservice.sycoca: Service type not found: "application/x-circuit"
kf5.kservice.sycoca: Service type not found: "application/x-microbe"
kf5.kservice.sycoca: Service type not found: "application/x-flowcode"
Shortcut for action  "file_new" "New File" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
qt5ct: D-Bus global menu: no
org.kde.ktechlab: no available sizes for  "C (.c)"
Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/libexec/kf5/klauncher'
kdeinit5: Launched KLauncher, pid = 5920, result = 0
qt5ct: using qt5ct plugin
Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
kdeinit5: opened connection to :0.0
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/file.so'
=================================================================
==5200==ERROR: AddressSanitizer: heap-use-after-free on address 0x607000468480 at pc 0x559eff8cce79 bp 0x7ffd17203380 sp 0x7ffd17203378
READ of size 8 at 0x607000468480 thread T0
    #0 0x559eff8cce78 in PropertyEditorItem::property() (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x18ee78)
    #1 0x559eff8c843c in PropertyEditor::updateDefaultsButton() (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x18a43c)
    #2 0x559eff8cae3c in PropertyEditor::resizeEvent(QResizeEvent*) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x18ce3c)
    #3 0x7f6471801ebd in QWidget::event(QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x19debd)
    #4 0x7f64718a4cdd in QFrame::event(QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x240cdd)
    #5 0x7f6471a161ea in QAbstractItemView::viewportEvent(QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3b21ea)
    #6 0x7f6470d3419a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x26a19a)
    #7 0x7f64717c34b0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x15f4b0)
    #8 0x7f64717ca96f in QApplication::notify(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16696f)
    #9 0x7f6470d34488 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x26a488)
    #10 0x7f64717f8e28 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x194e28)
    #11 0x7f64717f9bef in QWidget::setGeometry(QRect const&) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x195bef)
    #12 0x7f64718a5fb8 in QAbstractScrollAreaPrivate::layoutChildren_helper(bool*, bool*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x241fb8)
    #13 0x7f64718a6519 in QAbstractScrollAreaPrivate::layoutChildren() (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x242519)
    #14 0x7f64718a7851 in QAbstractScrollArea::event(QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x243851)
    #15 0x7f6471a15ec8 in QAbstractItemView::event(QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3b1ec8)
    #16 0x7f64717c34c0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x15f4c0)
    #17 0x7f64717ca96f in QApplication::notify(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16696f)
    #18 0x7f6470d34488 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x26a488)
    #19 0x7f64717f8e28 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x194e28)
    #20 0x7f64717f9bef in QWidget::setGeometry(QRect const&) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x195bef)
    #21 0x7f64717e2f4f in QWidgetItem::setGeometry(QRect const&) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x17ef4f)
    #22 0x7f64717d91af in QBoxLayout::setGeometry(QRect const&) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1751af)
[... cut ...]
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==5200==ABORTING
```

Perhaps a recursion on freeing objects somewhere?

If you need the build-developer.log let me know (only unused parameters).

Btw: this also happens with the package (0.50.0-2) from Debian bullseye: https://packages.debian.org/bullseye/ktechlab

zoltanp, 2020-12-15:
Thanks for the stack trace! The issue looks like a pointer-ownership related issue. Something got destroyed (a `Property`) and some pointer to it is not invalidated.

I've just reproduced the crash:
1. place and select a resistor
2. in the property editor click on any of its properties, in order to start editing it
3. click on the circuit, to stop editing of the properties
3.1. Expected: property value is kept, program continues
3.2. Actually: crash with the same stack trace from above

> Btw: this also happens with the package (0.50.0-2) from Debian bullseye: https://packages.debian.org/bullseye/ktechlab

As I see the situation, after fixing this issue, a new bugfix release should be done. 
I do not want to ship a crashing program.

abalfoort, 2020-12-16:
I'd like to see the change when you're done. I have been a programmer for the past 23 years but I've only recently begun studying on C.

zoltanp, 2020-12-18:
This issue should be fixed by:

https://invent.kde.org/sdk/ktechlab/-/commit/730a5492e4780504a80db1a27dc307d8f61b4696

https://github.com/ktechlab/ktechlab/commit/730a5492e4780504a80db1a27dc307d8f61b4696

Apparently it has been a dangling pointer :-)
In the fix I've went completely paranoic about the state of the widgets.

Please verify if the crash persists. It it is fixed, feel free to close this issue.

It's interesting that I couldn't reproduce the issue on Qt 5.12.8 / Ubuntu 20.04.

abalfoort, 2020-12-19:
I can confirm that this fixes the issue.
Thanks for the quick fix!

```
$ sh run-developer.sh
+ case "$(uname -s)" in
++ uname -s
+ echo Linux
Linux
+++ readlink -f run-developer.sh
++ dirname /home/solydxk/Downloads/ktechlab-master/run-developer.sh
+ SCRIPTDIR=/home/solydxk/Downloads/ktechlab-master
+ . /home/solydxk/Downloads/ktechlab-master/build-developer/prefix.sh
++ export PATH=/home/solydxk/Downloads/ktechlab-master/inst-developer/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
++ PATH=/home/solydxk/Downloads/ktechlab-master/inst-developer/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
++ export XDG_DATA_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-developer/share:/usr/share/xfce4:/usr/local/share/:/usr/share/:/usr/share
++ XDG_DATA_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-developer/share:/usr/share/xfce4:/usr/local/share/:/usr/share/:/usr/share
++ export XDG_CONFIG_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-developer/etc/xdg:/etc/xdg
++ XDG_CONFIG_DIRS=/home/solydxk/Downloads/ktechlab-master/inst-developer/etc/xdg:/etc/xdg
++ export QT_PLUGIN_PATH=/home/solydxk/Downloads/ktechlab-master/inst-developer/lib/x86_64-linux-gnu/plugins:
++ QT_PLUGIN_PATH=/home/solydxk/Downloads/ktechlab-master/inst-developer/lib/x86_64-linux-gnu/plugins:
++ export QML2_IMPORT_PATH=/home/solydxk/Downloads/ktechlab-master/inst-developer/lib/x86_64-linux-gnu/qml:
++ QML2_IMPORT_PATH=/home/solydxk/Downloads/ktechlab-master/inst-developer/lib/x86_64-linux-gnu/qml:
+ export ASAN_OPTIONS=new_delete_type_mismatch=0
+ ASAN_OPTIONS=new_delete_type_mismatch=0
+ export 'QT_MESSAGE_PATTERN=%{file} : %{line} %{type} %{function} -- %{message}'
+ QT_MESSAGE_PATTERN='%{file} : %{line} %{type} %{function} -- %{message}'
+ tee /home/solydxk/Downloads/ktechlab-master/developer-run.log
+ /home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab
unknown : 0 debug unknown -- using qt5ct plugin
unknown : 0 critical unknown -- Shortcut for action  "file_new" "New File" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
unknown : 0 debug unknown -- D-Bus global menu: no
unknown : 0 warning unknown -- no available sizes for  "C (.c)"
unknown : 0 critical unknown -- Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
unknown : 0 critical unknown -- Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
unknown : 0 critical unknown -- Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
unknown : 0 critical unknown -- Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
unknown : 0 warning unknown -- 0x6110004c97f0 deleted without having been removed from the factory first. This will leak standalone popupmenus and could lead to crashes.

=================================================================
==5188==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4608 byte(s) in 18 object(s) allocated from:
    #0 0x7f086b05c330 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xe9330)
    #1 0x7f0865e5a6d4  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x206d4)

Direct leak of 184 byte(s) in 1 object(s) allocated from:
    #0 0x7f086b05dd30 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xead30)
    #1 0x563ef916d3ef in Component::createResistance(Pin*, Pin*, double) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x5553ef)
    #2 0x563ef9169fe0 in Component::createResistance(ECNode*, ECNode*, double) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x551fe0)
    #3 0x563ef90c817f in Resistor::Resistor(ICNDocument*, bool, char const*) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x4b017f)
    #4 0x563ef90c78bb in Resistor::construct(ItemDocument*, bool, char const*) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x4af8bb)
    #5 0x563ef917a27a in ItemLibrary::createItem(QString const&, ItemDocument*, bool, char const*, bool) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x56227a)
    #6 0x563ef8fb5089 in ICNDocument::addItem(QString const&, QPoint const&, bool) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x39d089)
    #7 0x563ef8fa6154 in ItemView::dropEvent(QDropEvent*) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x38e154)
    #8 0x563ef8fab43b in CVBEditor::event(QEvent*) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x39343b)
    #9 0x7f0868a1b4c0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x15f4c0)

Direct leak of 184 byte(s) in 1 object(s) allocated from:
    #0 0x7f086b05dd30 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xead30)
    #1 0x563ef916d3ef in Component::createResistance(Pin*, Pin*, double) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x5553ef)
    #2 0x563ef9169fe0 in Component::createResistance(ECNode*, ECNode*, double) (/home/solydxk/Downloads/ktechlab-master/inst-developer/bin/ktechlab+0x551fe0)
[... cut ...]
Indirect leak of 45 byte(s) in 3 object(s) allocated from:
    #0 0x7f086b05c330 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xe9330)
    #1 0x7f08657a496e in _dbus_strdup (/lib/x86_64-linux-gnu/libdbus-1.so.3+0x3096e)

Indirect leak of 32 byte(s) in 2 object(s) allocated from:
    #0 0x7f086b05c720 in __interceptor_realloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xe9720)
    #1 0x7f08657a382b  (/lib/x86_64-linux-gnu/libdbus-1.so.3+0x2f82b)

SUMMARY: AddressSanitizer: 19842 byte(s) leaked in 165 allocation(s).
```

closed 2020-12-19 by abalfoort