ktechlab/ktechlab #61: Logic voltages are too low for MOSFETs and cannot be set higher

Tynach, 2021-04-02:
Logical inputs max out at 1.6 volts, but the MOSFETs seem to only switch state starting at around 2 volts.

While it would be nice if voltage could be configured on a per-input level, that is not the point of this bug. As it is, *none* of the MOSFET examples actually work because the input voltages are too low, and cannot be set higher (the number input in the logic settings simply doesn't allow the entry of values higher than 1.6).

Also, many *many* times I've encountered severely buggy behavior with MOSFETs in general, but I don't know how to reliably reproduce that problem, so I'll be filing a separate bug report for that if and when I figure out how to reproduce it.

zoltanp, 2021-04-02:
Hi,
thanks for reporting this issue.
I'm aware of at least 1 issue with MOSFETs, and yes, they are buggy:

https://invent.kde.org/sdk/ktechlab/-/issues/7

> the number input in the logic settings simply doesn't allow the entry of values higher than 1.6

This is unexpected and should definitely not work like that.

Tynach, 2021-04-09:
> This is unexpected and should definitely not work like that.

Well, it seems to be defined in the code [here](../blob/master/src/gui/logicwidget.ui#L176), and according to the 'blame' tool, it was added in the initial Qt 4 port... But digging deeper, a variant of that line was traced to [here](../blob/4a7d65127740c3ab7da3294ab721c39aec084e3e/src/gui/logicwidget.ui#L145), the initial svn to git import commit. As far as I can tell, 1.6 has always been the maximum allowed logic voltage.