ktechlab/ktechlab #62: Cannot build on Debian Stretch

elpibedeoro, 2021-04-20:
Hi,

I'm trying to build the latest version on Debian Stretch 64bit and I've got a few errors when I run make.

```
[ 27%] Building CXX object src/CMakeFiles/test_ktechlab.dir/gui/outputmethoddlg.cpp.o
/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp: In constructor ‘OutputMethodDlg::OutputMethodDlg(const QString&, const QUrl&, bool, QWidget*)’:
/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp:100:30: error: ‘class KUrlRequester’ has no member named ‘setAcceptMode’; did you mean ‘setAcceptDrops’?
     m_widget->outputFileURL->setAcceptMode(QFileDialog::AcceptSave);
                              ^~~~~~~~~~~~~
src/CMakeFiles/ktechlab.dir/build.make:890 : la recette pour la cible « src/CMakeFiles/ktechlab.dir/gui/outputmethoddlg.cpp.o » a échouée
make[2]: *** [src/CMakeFiles/ktechlab.dir/gui/outputmethoddlg.cpp.o] Erreur 1
CMakeFiles/Makefile2:241 : la recette pour la cible « src/CMakeFiles/ktechlab.dir/all » a échouée
make[1]: *** [src/CMakeFiles/ktechlab.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
[ 27%] Building CXX object src/CMakeFiles/test_ktechlab.dir/gui/scopeviewbase.cpp.o
In file included from /opt/build_ktechlab/ktechlab/src/gui/projectdlgs.cpp:32:0:
/opt/build_ktechlab/ktechlab/build/src/ui_processingoptionswidget.h: In member function ‘void Ui_ProcessingOptionsWidget::setupUi(QWidget*)’:
/opt/build_ktechlab/ktechlab/build/src/ui_processingoptionswidget.h:64:23: error: ‘class KUrlRequester’ has no member named ‘setAcceptMode’; did you mean ‘setAcceptDrops’?
         m_pOutputURL->setAcceptMode(QFileDialog::AcceptSave);
                       ^~~~~~~~~~~~~
src/CMakeFiles/test_ktechlab.dir/build.make:842 : la recette pour la cible « src/CMakeFiles/test_ktechlab.dir/gui/projectdlgs.cpp.o » a échouée
make[2]: *** [src/CMakeFiles/test_ktechlab.dir/gui/projectdlgs.cpp.o] Erreur 1
make[2]: *** Attente des tâches non terminées....
/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp: In constructor ‘OutputMethodDlg::OutputMethodDlg(const QString&, const QUrl&, bool, QWidget*)’:
/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp:100:30: error: ‘class KUrlRequester’ has no member named ‘setAcceptMode’; did you mean ‘setAcceptDrops’?
     m_widget->outputFileURL->setAcceptMode(QFileDialog::AcceptSave);
                              ^~~~~~~~~~~~~
src/CMakeFiles/test_ktechlab.dir/build.make:866 : la recette pour la cible « src/CMakeFiles/test_ktechlab.dir/gui/outputmethoddlg.cpp.o » a échouée
make[2]: *** [src/CMakeFiles/test_ktechlab.dir/gui/outputmethoddlg.cpp.o] Erreur 1
CMakeFiles/Makefile2:278 : la recette pour la cible « src/CMakeFiles/test_ktechlab.dir/all » a échouée
make[1]: *** [src/CMakeFiles/test_ktechlab.dir/all] Erreur 2
Makefile:138 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
```

These are my steps : 

`sudo apt install git build-essential cmake extra-cmake-modules gettext qt5-default libkf5declarative-dev libkf5khtml-dev libkf5doctools-dev libkf5texteditor-dev gpsim-dev glib2.0`
`cd /opt`
`git clone https://github.com/ktechlab/ktechlab.git`
`cd ktechlab`
`mkdir build && cd build`
`cmake ..`
`make -j4`

This is the result of cmake that it seems to be ok : 
```
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could not set up the appstream test. appstreamcli is missing.
-- Looking for __GLIBC__
-- Looking for __GLIBC__ - found
-- Performing Test _OFFT_IS_64BIT
-- Performing Test _OFFT_IS_64BIT - Success
-- Found KF5Completion: /usr/lib/x86_64-linux-gnu/cmake/KF5Completion/KF5CompletionConfig.cmake (found version "5.28.0") 
-- Found KF5Config: /usr/lib/x86_64-linux-gnu/cmake/KF5Config/KF5ConfigConfig.cmake (found version "5.28.0") 
-- Found KF5ConfigWidgets: /usr/lib/x86_64-linux-gnu/cmake/KF5ConfigWidgets/KF5ConfigWidgetsConfig.cmake (found version "5.28.0") 
[... 548 lines cut ...]
[ 27%] Building CXX object src/CMakeFiles/test_ktechlab.dir/gui/scopeviewbase.cpp.o
In file included from /opt/build_ktechlab/ktechlab/src/gui/projectdlgs.cpp:32:0:
/opt/build_ktechlab/ktechlab/build/src/ui_processingoptionswidget.h: In member function ‘void Ui_ProcessingOptionsWidget::setupUi(QWidget*)’:
/opt/build_ktechlab/ktechlab/build/src/ui_processingoptionswidget.h:64:23: error: ‘class KUrlRequester’ has no member named ‘setAcceptMode’; did you mean ‘setAcceptDrops’?
         m_pOutputURL->setAcceptMode(QFileDialog::AcceptSave);
                       ^~~~~~~~~~~~~
src/CMakeFiles/test_ktechlab.dir/build.make:842 : la recette pour la cible « src/CMakeFiles/test_ktechlab.dir/gui/projectdlgs.cpp.o » a échouée
make[2]: *** [src/CMakeFiles/test_ktechlab.dir/gui/projectdlgs.cpp.o] Erreur 1
make[2]: *** Attente des tâches non terminées....
/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp: In constructor ‘OutputMethodDlg::OutputMethodDlg(const QString&, const QUrl&, bool, QWidget*)’:
/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp:100:30: error: ‘class KUrlRequester’ has no member named ‘setAcceptMode’; did you mean ‘setAcceptDrops’?
     m_widget->outputFileURL->setAcceptMode(QFileDialog::AcceptSave);
                              ^~~~~~~~~~~~~
src/CMakeFiles/test_ktechlab.dir/build.make:866 : la recette pour la cible « src/CMakeFiles/test_ktechlab.dir/gui/outputmethoddlg.cpp.o » a échouée
make[2]: *** [src/CMakeFiles/test_ktechlab.dir/gui/outputmethoddlg.cpp.o] Erreur 1
CMakeFiles/Makefile2:278 : la recette pour la cible « src/CMakeFiles/test_ktechlab.dir/all » a échouée
make[1]: *** [src/CMakeFiles/test_ktechlab.dir/all] Erreur 2
Makefile:138 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
```

I've got a few errors if I launch build-simple.sh too. If it worths I could show its result.

Do you have any idea about this issue ? Did I forget anything maybe ?
Thanks in advance for your help.

zoltanp, 2021-04-20:
Hi,
based on the error, it looks like the KDE Frameworks included in Debian
Stretch are too old and they are not compatible with latest KTechLab.

/opt/build_ktechlab/ktechlab/src/gui/outputmethoddlg.cpp:100:30:
error: ‘class KUrlRequester’ has no member named ‘setAcceptMode’; did
you mean ‘setAcceptDrops’?

KUrlRequester::setAcceptMode() has been introduced in version 5.33 of KDE
Framework KIO:
https://api.kde.org/frameworks-api/frameworks-apidocs/frameworks/kio/html/classKUrlRequester.html#ad3ad89008d0ea59335d7495084a57fe9

If possible, I recommend using a newer Linux distribution. Alternatively,
you might try to install newer KDE Frameworks, but I assume that is not
trivial.

Probably some all-in-one application packagee would be useful now
(Flatpack, Appimage, ...), but as far as I know currently there is none
readily available.

elpibedeoro, 2021-04-21:
Hi,

A big thank you for your help and your reactivity !!
I upgrade to Debian 10 and hope it will works.

Have a nice day.

elpibedeoro, 2021-04-27:
Hi,

Everything works as well under Debian 10.
Thank you so much for your help ! 

This can be closed.

Have a nice day.

closed 2021-04-28 by zoltanp