ktechlab/ktechlab #65: from 12 v dc battery connected to 2 resistors in parallel.

noisecode3, 2021-11-25:
Hi, when I connect 12 v dc battery connected to 2 resistors in parallel I get junk result. I says 0 V on the left side and 5 V on the right side and its 1 mA across the whole circuit. I'm studding basics of circuit calculations in school. I'm sure that should not happen. and also the button for changing the volt of the battery disappears sometime and it usually crashes after some time. I build and executed run-simple.sh this is the output. 

`+ case "$(uname -s)" in
++ uname -s
+ echo Linux
Linux
+++ readlink -f ./run-simple.sh
++ dirname /home/marbangens/Documents/Code/ktechlab/run-simple.sh
+ SCRIPTDIR=/home/marbangens/Documents/Code/ktechlab
+ . /home/marbangens/Documents/Code/ktechlab/build-simple/prefix.sh
++ export PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/bin:/usr/local/bin:/usr/bin:/bin:/usr/games:/usr/lib64/libexec/kf5:/usr/lib64/qt5/bin
++ PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/bin:/usr/local/bin:/usr/bin:/bin:/usr/games:/usr/lib64/libexec/kf5:/usr/lib64/qt5/bin
++ export XDG_DATA_DIRS=/home/marbangens/Documents/Code/ktechlab/inst-simple/share:/usr/local/share/:/usr/share/
++ XDG_DATA_DIRS=/home/marbangens/Documents/Code/ktechlab/inst-simple/share:/usr/local/share/:/usr/share/
++ export XDG_CONFIG_DIRS=/home/marbangens/Documents/Code/ktechlab/inst-simple/etc/xdg:/home/marbangens/.config/kdedefaults:/etc/xdg:/etc/kde/xdg
++ XDG_CONFIG_DIRS=/home/marbangens/Documents/Code/ktechlab/inst-simple/etc/xdg:/home/marbangens/.config/kdedefaults:/etc/xdg:/etc/kde/xdg
++ export QT_PLUGIN_PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/lib64/plugins:
++ QT_PLUGIN_PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/lib64/plugins:
++ export QML2_IMPORT_PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/lib64/qml:
++ QML2_IMPORT_PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/lib64/qml:
++ export QT_QUICK_CONTROLS_STYLE_PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/lib64/qml/QtQuick/Controls.2/:
++ QT_QUICK_CONTROLS_STYLE_PATH=/home/marbangens/Documents/Code/ktechlab/inst-simple/lib64/qml/QtQuick/Controls.2/:
+ /home/marbangens/Documents/Code/ktechlab/inst-simple/bin/ktechlab
kf.xmlgui: Shortcut for action  "file_new" "New File" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
kf.xmlgui: 0x172f7a0 deleted without having been removed from the factory first. This will leak standalone popupmenus and could lead to crashes.
org.kde.ktechlab: Wire voltage error:  -5
org.kde.ktechlab: Wire voltage error:  -5
org.kde.ktechlab: Wire voltage error:  -5
org.kde.ktechlab: Wire voltage error:  -5
kf.xmlgui: 0x18ae2a0 deleted without having been removed from the factory first. This will leak standalone popupmenus and could lead to crashes.
kf.notifications: Playing audio notification failed: Destroyed
`

noisecode3, 2021-11-26:
Okej I think I read those values wrong sorry, but there some bug in the gui.

noisecode3, 2021-12-03:
Im on slackware current.. I can'r reproduce this in xfce4 only in kde5. I guess its fine.. Wire voltage error: -5 seems harmless.

zoltanp, 2022-06-26:
I cannot reproduce this. Maybe the simulator has been paused ?

zoltanp, 2022-06-26:
Closing, feel free to reopen if this issue happens again.
There is an open issue about texts of the components, maybe that has been the original cause ?
https://bugs.kde.org/show_bug.cgi?id=455963

closed 2022-06-26 by zoltanp