ktechlab/ktechlab #66: Unable to compile KTechLab '‘class KUrlRequester’ has no member named ‘setAcceptMode’'

TCH68k, 2022-01-12:
I simply ran the `./build-simple.sh` command after i installed all of the dependencies. The OS is Debian Stretch. The log is here: http://oscomp.hu/depot/kdetechlab.log

zoltanp, 2022-01-12:
Hi,

apparently the used KDE libraries are too old to compile KTechLab out-of-the box. I see you have version 5.28 (from the log file) while for the `setAcceptMode` method at least 5.33 is required [1].

For compiling KTechLab I see two options:
- Try to update the KDE libraries if this is feasible
- Try patch out the calls to this method, maybe this is the only one causing compilation failure. See [2][3][4]. Possibly this is the only API used which is not available in KF5 version 5.28. 

Hope this is useful.

[1] https://api.kde.org/frameworks-api/frameworks-apidocs/frameworks/kio/html/classKUrlRequester.html#ad3ad89008d0ea59335d7495084a57fe9

[2] https://invent.kde.org/sdk/ktechlab/-/blob/master/src/gui/outputmethoddlg.cpp#L100

[3] https://invent.kde.org/sdk/ktechlab/-/blob/master/src/gui/processingoptionswidget.ui#L61

[4] https://invent.kde.org/sdk/ktechlab/-/blob/master/src/gui/imageexportdlg.cpp#L48

TCH68k, 2022-01-12:
Thanks, i think i'll make a wrapper function which'll "emulate" `setAcceptMode` and patch KTechLab.