ktechlab/ktechlab #70: LC generators with inductive coupling how to create?

Nikita-Presnov, 2022-02-07:
LC generators with inductive coupling how to create? Or transformer?
I have not found the element responsible for creating mutual induction anywhere.
It is also not very clear if there is an oscilloscope function somewhere, I want to see the same sine from a simple circuit.

Some icons are hard to see with dark theme.

The program needs to be brought to a release state.

zoltanp, 2022-02-08:
Hi,
inductive coupling is not implemented. You might try to model it with current-controlled-current sources or similar...

To see a sine wave, connect a Voltage probe or Current probe to the location of interest and on the Oscilloscope view (not the Scope screen) you should see waveforms.

About icons and dark themes: this is valid and probably worth a separate issue for tracking its status.

About release state: what do you mean specifically?

zoltanp, 2022-06-26:
Cross-linking with KDE bugzilla:

https://bugs.kde.org/show_bug.cgi?id=455964