ktechlab/ktechlab #73: The "Open" dialog doesn't show any files when "*|All Files" is chosen as selector

yurivict, 2022-07-08:
The "Open" dialog shows only directories even though there are many files there too, including *.circuit files.

Choosing the "*.circuit" selector allows to show such files.

Version: 0.50.0
FreeBSD 13.1

zoltanp, 2022-07-10:
Thanks for reporting the issues. I wonder what would be more useful: to fix the issues in a bugfix release for version 0.50.0, or fix-up the git master and make a release from that (currently there are some regressions in git master). How do you see this?

I have fixed this issue in git master:

https://invent.kde.org/sdk/ktechlab/-/commit/46c30e122135c840c8b4f9eb92c0b16a89a839cf

https://github.com/ktechlab/ktechlab/commit/46c30e122135c840c8b4f9eb92c0b16a89a839cf

yurivict, 2022-07-12:
Bugfix release seems better because it's important to keep stability of releases.

zoltanp, 2022-08-09:
Cross-linking with KDE bugzilla : https://bugs.kde.org/show_bug.cgi?id=457665

closed 2025-11-22 by zoltanp