ktechlab/ktechlab #75: Program crash (segmentation fault) when changing component value

ibexnet, 2022-09-12:
Hi everybody,
The program crash when I change resistor value but sometime also with others components.
I notice that the the behavior occur only when change the value from _ItemEditor_ menu (on right bar).
If I change the value on the top bar everything work fine (but it is useless for others device because appear only few adjustable values).

regards

Info about the environment:
- KTechLab Version 0.50.0
- OS Ubuntu 22.04.1 LTS
- Kernel Linux 5.15.0-47-generic

zoltanp, 2022-09-13:
Hi,

do you have a specific sequence of actions that make KTechLab crash? I will try to reproduce the crash, in order to investigate.

Also you have installed KTechLab from package manager or compiled it from sources? It would be nice to get a useful backtrace / crash dump, but for that at least the symbols, or preferably the debug information is needed for the ktechlab executable. Packages by default do not contain debug information.

Best regards,

 Zoltan

ibexnet, 2022-09-14:
Hi Zotlan, thanks for the reply.
I installed the KTechLab from package manager (snap store).

The sequence of action for reproduce the crash are:

1. Create or open a circuit
2. add a resistor (seem that not all components are impacted by this issue, I just tried with resistor, capacitor and battery, for example the changes parameters for transistor not make crash the program)
3. click on the resistor and change resistance value from _Item Editor_ 
4. then click on the circuit pane for example for select another component ( if remain with the mouse on the _Item Editor_ pane the program don't crash)
5. few second after apply new resistance value the program crash

For debug this is what happen if you launch the program from terminal:
``
Starting logging to  "/tmp/ktechlab-pid-403826-log"
(II) Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
(WW) KServiceTypeTrader: serviceType "QImageIOPlugins" not found
(Critical) Shortcut for action  "file_new" "New File" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "cancelCurrentOperation" "Cancel" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "edit_delete" "Delete" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "edit_raise" "Raise Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
(Critical) Shortcut for action  "edit_lower" "Lower Selection" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[1]    403826 segmentation fault (core dumped)  ktechlab
``

I had try to use _strace_ to see if we can catch some useful info but I don't find nothing relevant, anyway I attached the output maybe is useful for you 

[ktechlab.ZIP](https://github.com/ktechlab/ktechlab/files/9567260/ktechlab.ZIP)

zoltanp, 2022-09-14:
This crash looks very similar to me to this one:

https://github.com/ktechlab/ktechlab/issues/60#issuecomment-745584243

.. which has been fixed in git master, but the fix did not make it into any release yet:

https://invent.kde.org/sdk/ktechlab/-/commit/730a5492e4780504a80db1a27dc307d8f61b4696

I wonder how would be easy to test the latest git master code, where hopefully this issue is fixed... One option would be to compile KTechLab from sources, but I assume that is not a very convenient thing to do ...

ibexnet, 2022-09-14:
OK thanks, I will try to build the source and I see if the issue is fixed

zoltanp, 2022-09-16:
I've verified git master on Ubuntu 22.04 and it doesn't crash. So I assume that it has been fixed in git master / duplicate of #60 . At least until shown otherwise.

zoltanp, 2025-11-22:
Assuming fixed, closing. In case you have issues, feel free to signal them.

closed 2025-11-22 by zoltanp