ktechlab/ktechlab #77: AC voltage

artenax, 2023-12-30:
I have been interested in this question for a long time after Electronic Workbench / Multisim. How can I view the AC voltage in a voltmeter? (RMS). The arrow is constantly jumping.

artenax, 2024-01-13:
In general, if you reduce the fps to a minimum in the settings (10 fps), press F10 (Simulation) and hover the mouse over the wire, it will show the current voltage and amperage. However, in the case of low Hz, these values will drop over time. You need to restart the program, press F10 and quickly look at the values. While the voltmeter and ammeter on the schematic will show nonsense, don't pay attention.

zoltanp, 2024-01-16:
> I have been interested in this question for a long time after Electronic Workbench / Multisim. How can I view the AC voltage in a voltmeter? (RMS). The arrow is constantly jumping.

Currently it is not possible to directly display the RMS value of a signal, only its instantaneous value.

As a suggestion, maybe you could create a circuit (and reuse it as a sub-circuit) that transforms a voltage signal into a voltage corresponding to the RMS value of the signal.

For example, see the above circuit that rectifies its input and applies down-pass filtering:

[image]

With some adequate tuning of the filter, you should get a decent RMS value display (plus-minus various imperfections of the components)

artenax, 2024-01-21:
I've tried this. Only at first I didn't use a resistor and connected the capacitor directly as in real voltage rectifier circuits and the voltmeter readings were unstable.

I added a resistor and the voltmeter readings became pretty stable. I also added a ground after the rectifier. But this is not the RMS. It is rectifying Peak with an underestimation of about 1.035 times. We should also use online converters to convert to RMS or do all calculations in Peak, assuming Peak = RMS.
https://calculator.academy/peak-current-calculator/
https://www.allaboutcircuits.com/tools/rms-voltage-calculator/

But it is much easier to see the values when hovering the mouse.

zoltanp, 2024-01-27:
I haven't done  the calculations, so that circuit above might not really give the RMS value.

I just want to point out that calculating the RMS value of a non-periodic signal is a non-trivial problem, because one has to select the time interval on which to perform the calculation of the mean value (of the squared signal).

So for implementing RMS display in an AC voltmeter, that works with arbitrary input, the component would need a time interval parameter. And the displayed RMS would vary in time, if the input signal is not periodic or the time interval of the calculations is too short.